PX4/PX4-Autopilot #3997: Excessive offset for mag 0

brendanogrady1, 2016-03-13:
cal] ERROR: Excessive offset for mag 0
[cal] calibration failed: mag
[PX4 sensor calibration txt.txt](https://github.com/PX4/Firmware/files/170794/PX4.sensor.calibration.txt.txt)

kd0aij, 2016-03-13:
try latest master (see 8e4a1ca)
note that your offsets didn't get saved in the list you posted

kd0aij, 2016-03-13:
also, make sure to minimize stray magnetic fields influencing the magnetometer, or it won't work well

Inspirati, 2016-03-13:
pls note: i found i had to import your patch in order to have my work on
the new LIS3MDL driver pass the calibration routines expectations too..
(excess offsets)

Inspirati, 2016-03-13:
thanks Mark, i found while just now doing development testing, that the
lis2mdl device would never report negative z-axis values, (at first thought
about bus corruption problems), until i waved the device over my laptop
computer!), whaalaay, problem solved and i successful moved onto completing
calibration..//

brendanogrady1, 2016-03-13:
Hi, I am new to everything here. I do not know where to find the offsets output in the QGroundControl program. I found this issues forum by searching the error on google and joined GitHub to post my own issue. I'm in a bit over my head but trying to learn fast.

I re-ran sensor calibration with an extra long cable connected to my computer and did it as far from anything as I could get, but got the same result.

I ran analyze and turned on everything... not sure if this will help?
[analyze log.txt](https://github.com/PX4/Firmware/files/171179/analyze.log.txt)

[analyze log_compressed.txt](https://github.com/PX4/Firmware/files/171178/analyze.log_compressed.txt)

Thank you and I hope I am not wasting your time.

LorenzMeier, 2016-03-13:
@brendanogrady1 Try to flash the "developer version" (tick the "Advanced checkbox at the bottom of the flash widget"). That should fix it.

brendanogrady1, 2016-03-13:
Well, that worked! Thank you so much!

Awesome!

closed 2016-03-13 by LorenzMeier

brendanogrady1, 2016-03-13:
Info shows a warning but calibration worked:

offsets: 0.4796, 0.2997, -1.0175, #0