PX4/PX4-Autopilot #4001: Voltage indication wrong for Pixracer when running with ACSP4

LorenzMeier, 2016-03-13:
@kd0aij It would be great if you could do some bench tests on the voltage output and adjust the scaling value here: https://github.com/PX4/Firmware/blob/master/src/modules/sensors/sensors.cpp#L868

kd0aij, 2016-03-13:
rechecked the scale factor. My test equipment says the voltage reading has better than 1% accuracy

timoski, 2016-04-02:
There is a bug at line 929. https://github.com/PX4/Firmware/blob/master/src/modules/sensors/sensors.cpp#L929.
It sets the scaling for the voltage but should be the scaling for the current!

kd0aij, 2016-04-02:
That bug would only be exposed if the configuration were not for PX4FMUxxx or AEROCORE. And then only if the current scaling parameter were negative. It's an easy one to fix though; a PR would have been less effort than writing this comment :)

kd0aij, 2016-04-02:
https://github.com/PX4/Firmware/pull/4143

julianoes, 2016-04-04:
Can now be closed, right?

closed 2016-04-04 by kd0aij