PX4/PX4-Autopilot #4004: sdlog2 poor performance

kd0aij, 2016-03-14:
This is a snapshot from a user flight testing acro mode on PixRacer. The sample rate is very irregular and there is insufficient data to analyze performance. 

[image]
http://logs.uaventure.com/view/zgccLTJ22PQcsr6mtbAMgg

tumbili, 2016-03-14:
@kd0aij I just did a flight with latest master today in the morning and logging was fine. I assume this is related to the SD card.

kd0aij, 2016-03-14:
This user also reports: "QGC reported several disconnects-reconnects too, which was completely unexpected with a USB connection."
The log shows STAT.Load = 0.41, and regarding rate tracking, see his comments (second flight) here: http://discuss.px4.io/t/pixraxer-as-a-racer/183/27?u=mark_whitehorn

dagar, 2016-03-14:
Do you have the postflight perf file?

kd0aij, 2016-03-14:
@dagar 
```
PERFORMANCE COUNTERS POST-FLIGHT

sd write: 6011 events, 0 overruns, 358448084us elapsed, 59632us avg, min 788us max 514514us 69108.883us rms
navigator: 41375 events, 0 overruns, 2831327us elapsed, 68us avg, min 29us max 886us 1252.540us rms
mc_att_control: 124127 events, 0 overruns, 7167871us elapsed, 57us avg, min 34us max 285us 24.572us rms
mavlink_txe: 1 events
mavlink_el: 44768 events, 0 overruns, 3520492us elapsed, 78us avg, min 34us max 2084us 139.940us rms
mavlink_txe: 0 events
mavlink_el: 49300 events, 0 overruns, 4078687us elapsed, 82us avg, min 43us max 2227us 148.619us rms
mavlink_txe: 0 events
mavlink_el: 242550 events, 0 overruns, 32750271us elapsed, 135us avg, min 48us max 639869us 3618.694us rms
sensor task update: 124129 events, 0 overruns, 10807238us elapsed, 87us avg, min 46us max 1120us 56.587us rms
mpu9250_duplicates: 125039 events
mpu9250_reset_retries: 0 events
mpu9250_good_transfers: 497650 events
mpu9250_bad_registers: 0 events
mpu9250_bad_transfers: 0 events
mpu9250_read: 622692 events, 0 overruns, 41198862us elapsed, 66us avg, min 39us max 108us 20.883us rms
mpu9250_gyro_read: 0 events
mpu9250_accel_read: 0 events
mpu9250_mag_read: 0 events
ctrl_latency: 124132 events, 1 overruns, 73211372us elapsed, 589us avg, min 12us max 1962us 189.288us rms
mpu6000_duplicates: 124133 events
mpu6000_reset_retries: 0 events
mpu6000_good_transfers: 498570 events
mpu6000_bad_registers: 0 events
mpu6000_bad_transfers: 0 events
mpu6000_read: 622705 events, 0 overruns, 30930699us elapsed, 49us avg, min 24us max 71us 13.174us rms
mpu6000_gyro_read: 0 events
mpu6000_accel_read: 0 events
hmc5883_conf_errors: 0 events
hmc5883_range_errors: 0 events
hmc5883_buffer_overflows: 23279 events
hmc5883_comms_errors: 162 events
hmc5883_read: 23441 events, 0 overruns, 839472us elapsed, 35us avg, min 14us max 1529us 78.965us rms
hmc5883_conf_errors: 0 events
hmc5883_range_errors: 0 events
hmc5883_buffer_overflows: 56690 events
hmc5883_comms_errors: 0 events
hmc5883_read: 56690 events, 0 overruns, 62023531us elapsed, 1094us avg, min 872us max 3671us 340.163us rms
adc_samples: 448362 events, 0 overruns, 1213426us elapsed, 2us avg, min 2us max 3us 0.499us rms
ms5611_buffer_overflows: 35406 events
ms5611_comms_errors: 0 events
ms5611_measure: 47207 events, 0 overruns, 416798us elapsed, 8us avg, min 5us max 1600us 40.704us rms
ms5611_read: 47207 events, 0 overruns, 4576774us elapsed, 96us avg, min 10us max 2110us 138.601us rms
DMA allocations: 1 events

LOAD POST-FLIGHT

Processes: 22 total, 2 running, 20 sleeping
CPU usage: 32.43% tasks, 6.56% sched, 61.01% idle
Uptime: 605.154s total, 347.398s idle

 PID COMMAND                   CPU(ms) CPU(%)  USED/STACK PRIO(BASE) STATE 
   0 Idle Task                  347398 61.012     0/    0   0 (  0)  READY 
   1 hpwork                      33499  5.154   852/ 1592 192 (192)  w:sig 
   2 lpwork                       2281  0.281   356/ 1592  50 ( 50)  READY 
   3 init                         1807  0.000  1316/ 2496 100 (100)  w:sem 
  84 dataman                         2  0.000   684/ 1192  90 ( 90)  w:sem 
 103 sensors                     18542  2.811  1700/ 1992 250 (250)  w:sem 
 105 gps                          1064  0.093   732/ 1192 220 (220)  w:sem 
 107 commander                    3259  0.468  3324/ 3592 140 (140)  w:sig 
 108 commander_low_prio             24  0.000   628/ 2872  50 ( 50)  w:sem 
 114 mavlink_if0                 35539  5.529  2444/ 2696 100 (100)  w:sig 
 115 mavlink_rcv_if0                47  0.000   828/ 2096 175 (175)  w:sem 
 118 mavlink_if1                  5467  0.749  2380/ 2696 100 (100)  w:sig 
 119 mavlink_rcv_if1                49  0.000   828/ 2096 175 (175)  w:sem 
 125 mavlink_if2                  4855  0.749  2380/ 2696 100 (100)  w:sig 
 126 mavlink_rcv_if2               905  0.093  1116/ 2096 175 (175)  w:sem 
 149 sdlog2                      41651  0.000  3052/ 3392  70 ( 70)  RUN   
 168 attitude_estimator_q        46678  7.403  2076/ 2496 250 (250)  w:sem 
 173 position_estimator_inav     21344  3.373  4844/ 5296 250 (250)  w:sem 
 175 mc_att_control              13556  2.343  1148/ 1496 250 (250)  w:sem 
 177 mc_pos_control               7431  0.843  1476/ 1896 250 (250)  w:sem 
 186 navigator                    5075  0.656   940/ 1496 105 (105)  w:sem 
 197 frsky_telemetry             12525  1.874   676/ 1096 200 (200)  w:sem
```

dagar, 2016-03-14:
Your sd write looks terrible.

    sd write: 6011 events, 0 overruns, 358448084us elapsed, 59632us avg, min 788us max 514514us 69108.883us rms

For reference this is what I have from a recent fixed wing flight on a pixhawk.

    sd write: 51200 events, 0 overruns, 90219920us elapsed, 1762us avg, min 9us max 76776us 2679.665us rms

**59632us** avg vs 1762us avg

dagar, 2016-03-14:
The perf counter is around the write.
https://github.com/PX4/Firmware/blob/master/src/modules/sdlog2/sdlog2.c#L649

kd0aij, 2016-03-14:
Assuming the DSYNC flag on the open makes this a blocking call, that would mean the SD card is taking too long for each physical write, and either the card is or its filesystem is too slow. 
How frequently does sdlog2 call fsync?

dagar, 2016-03-14:
It's just below. Every 10 writes.
https://github.com/PX4/Firmware/blob/master/src/modules/sdlog2/sdlog2.c#L676

kd0aij, 2016-03-14:
I can't find adequate documentation of the O_SYNC and O_DSYNC flags for NuttX, but it sounds like the effect is equivalent to fsync on every write. That would not be good.

kd0aij, 2016-03-17:
@dagar I tried removing the O_DSYNC flag from the logfile fopen() call, and there was no obvious effect on the perf_counter results.  I have no idea what it might be doing that wouldn't show in the perf_counter. Can you point me to any proper documentation for it?

dagar, 2016-03-17:
Isn't it more likely an SD card problem? Aren't the flags and SD identical on the pixhawk? Do you have an earlier log (this pixracer + sdcard) where this wasn't a problem?

I'll have a pixracer in a 250 quad soon. I'll take a look at the log quality and perf after my first flight.

kd0aij, 2016-03-17:
I just meant that removing the flag didn't affect performance (of my system) in any obvious way. My logs have never had large dropouts like the user's (olex) log. With no documentation on the NuttX fopen, I'm not inclined to proceed further with this.

kd0aij, 2016-04-01:
validating SD card writes as part of https://github.com/PX4/Firmware/pull/4117

closed 2016-04-01 by kd0aij