PX4/PX4-Autopilot #4005: Dual GPS support

jcooper21, 2016-03-14:
It would be nice to have dual GPS support with px4.

LorenzMeier, 2016-03-14:
@jcooper21 Could you explain why you need two GPS units? Just because others have that functionality, or can you make an actual flight performance argument?

tubeme, 2016-03-14:
@LorenzMeier It might be a redundancy concept. But I rather appreciate it if we use UAVCAN then have a 2 GPS + 3 Pixhawk + 3 airpseed sensors all connected on UAVCAN for redundancy purposes. Otherwise it is not worth it.

jcooper21, 2016-03-14:
@LorenzMeier it is for redundancy and commercial application. For commercial use, designs with maximum redundancy will always be preferred. Using UAVCAN for that should work well I think.

suraqis, 2016-03-21:
Dual GNSS for redundancy is wanted feature, but thinking strictly I don't know a case when one GNSS failed and second saves. In case of jamming or interference it's likely to fail for both receivers. But other opportunity - dual GNSS heading. It could be cool to have compass redundancy via dual GNSS heading, but as far as I understand it should be done in a RTK like way, otherwise it will be inaccurate.

GervaisK, 2016-03-23:
I´m in need for that too. Dual GNSS support (at least for the logs) allows hardware comparisons. An easy way to find out about GNSS assy quality.

tumbili, 2016-05-03:
See https://github.com/PX4/Firmware/pull/4301

closed 2016-05-03 by tumbili