PX4/PX4-Autopilot #4007: MC: Support for retracts

tops4u, 2016-03-14:
AFAIK electric retracts are currently not supported for Multicopters. When Starting or Landing in Auto Mode (Mission or while RTL) support for retracts would be desirable. I.e. if RTL is issued because of RC Problems the pilot my not be able to lower Retracts while RTL. Same goes for Mission Mode where the Retracts can not be programmed in the Mission. 

Proposal to make an AUX Output assignable to support retracts (as it is currently done for Gimbal or Shutter control). AUTO Modes would then control this Output after takeoff and prior to landing.

marcmarcet, 2016-05-27:
Hi,

I'm pretty new to pixhawk, but just to confim, there is no way to setup retracts? (I'm using qgroundcontrol) for example, if I conect them to AUX 4? I have managed to configure them this way with APM, but I'm trying to do it now with PX4

tops4u, 2016-05-28:
See above Issue #4276 , this is foreseen for Version 1.5 of PX 4.

closed 2016-05-28 by tops4u