PX4/PX4-Autopilot #4017: FW auto land / MAV_CMD_DO_LAND_START

dagar, 2016-03-15:
@LorenzMeier @AndreasAntener @DonLakeFlyer @sanderux 

I'm looking for input to improve fixed wing auto land as part of return to land.

**Questions**
 1. Would the proper way to do this be with MAV_CMD_DO_LAND_START marking the portion of the mission that auto lands?

 2. Could we require a proper landing setup, perhaps if RTL_LAND_DELAY is not disabled?

 3. Would this also work for the auto land mode in fixed wing?

kd0aij, 2016-03-15:
Does "return to land" imply "return to launchpoint, then land", or just "return to launch"?

dagar, 2016-03-15:
1. climb to RTL_RETURN_ALT
 2. return to home position
 3. descend to RTL_DESCEND_ALT
 4. loiter for RTL_LAND_DELAY (indefinite if -1)
 5. land at home

jcooper21, 2016-03-30:
if you just entered the planes glide slope then it could back calculate the altitude which it should start to decend to the home point.

dagar, 2016-03-30:
@jcooper21 yes I've been thinking about adding that to landing. Something like spiral descend and break  out at the glide slope intersection.

jcooper21, 2016-03-30:
Yes those would be powerful features and also my favourite if implemented.

sanderux, 2016-03-31:
@dagar this might be helpful: https://github.com/PX4/Firmware/blob/master/src/modules/fw_pos_control_l1/fw_pos_control_l1_main.cpp#L2275-L2278

LorenzMeier, 2016-12-28:
Fixed.

closed 2016-12-28 by LorenzMeier