PX4/PX4-Autopilot #4024: How to set waypoint with the position caculated by vision

jian-li, 2016-03-16:
Hi, I am new to pixhawk. I successfully intergrated onboard rgbd vo and pixhawk, the x, y axis value in the LOCAL_POSITION_NED is correct in the qgroundcontrol, but the z axis value is negative, is it correct? Plus,  I don't sure weather I can set the waypoint using the qgroudcontrol, if I use the qgroundcontrol to set waypoint, how the altitude should be set?Thx~

liang-tang, 2016-03-21:
```ruby
// This is an example Custom Command Qml file. You have full access to the entire Qml language
// for creating any user interface you like. From the ui you can affect the following changes
// with respect to your vehicle:
//    1) Sending COMMAND_LONG commands out over mavlink using QGCButton control
//    2) Modifying parameters
//
// When developing custom Qml file implementations. You must restart QGroundControl to pick up
// the changes. You need to do this even if you select Clear Qml file. Not sure what at the this
// point. Qt must be caching the files somewhere.

import QtQuick 2.2

import QGroundControl.Controls 1.0
import QGroundControl.FactSystem 1.0
import QGroundControl.FactControls 1.0
import QGroundControl.Controllers 1.0

FactPanel {
    id: panel

    property var qgcView: null // Temporary hack for broken QGC parameter validation implementation

    CustomCommandWidgetController { id: controller; factPanel: panel }

    // Your own custom changes start here - everything else above is always required

    Column {
        // The QGCButton control is provided by QGroundControl.Controls. It is a wrapper around
        // the standard Qml Button element which using the default QGC font and color palette.

        QGCButton {
            text: "waypoint1"
            onClicked: controller.sendCommand(16, 50, 0, 1, 1, 1, 1, 30.5454457, 104.0725851, 600)
        }

        QGCButton {
            text: "waypoint2"
            onClicked: controller.sendCommand(16, 50, 0, 1, 1, 1, 1, 55.7533684, 37.6287914, 600)
        }

        QGCButton {
            text: "waypoint3"
            onClicked: controller.sendCommand(16, 50, 0, 1, 1, 1, 1, 55.7524809, 37.6273322, 600)
        }

        // The FactTextField control is provides by GroundControl.FactControls. It is a wrapper
        // around the Qml TextField element which allows you to bind it directly to any parameter.
       // The parameter is changed automatically when you click enter or click away from the field.
        // Understand that there is currently no value validation. So you may crash your vehicle by
        // setting a parameter to an incorrect value. Validation will come in the future.

        // Be very careful when referencing parameters. If you specify a parameter which does not exist
        // QGroundControl will warn and shutdown.

        FactTextField {
            // The -1 signals default component id.
            // You can replace it with a specific component id if you like
            fact: controller.getParameterFact(-1, "MAV_SYS_ID")
        }
    }

    // Your own custom changes end here - everything else below is always required
}
```

                                                                                                                    62,1          38%

jian-li, 2016-03-21:
Thank you @liang-tang ! I follow the steps in http://dev.px4.io/advanced-fake-gps.html, and in qgroundcontrol I get the "3d lock" message, but the light don't light in green color!

liang-tang, 2016-03-21:
The meaning of  "3d lock" message is that gps is working well, and the light isn't green maybe there 's other reasons.

jian-li, 2016-03-21:
Ok! Another problem is that if I set the waypoint in qgc or in mavros, which axis should I choose(NED for waypoint vs mavros ENU?)?

liang-tang, 2016-03-21:
NED probably.

closed 2016-11-08 by dagar