PX4/PX4-Autopilot #4025: position estimator inav gives garbage z velocity with mocap data

holmeswk, 2016-03-17:
Passing dummy mocap data with no motion to a level stationary pixhawk results in non-zero velocity estimates. Ive seen as much as 1 m/s. I was using a previous build and had no trouble. Ive looked through the code but cant find any obvious changes. It is likely i was running another module before that was correcting for imu bias that was running alongside the position_inav node.

holmeswk, 2016-03-17:
i killed the baro weight and cracked up the imu bias weight. this gave me reasonable velocities.

jgoppert, 2016-03-19:
@holmeswk We are transitioning from INAV to LPE. Try using this branch: https://github.com/PX4/Firmware/pull/4036
This is a standard kalman filter instead of a steady state kalman filter like inav, so you can just set the standard deviation of the mocap measurements, and it will do the rest.

jgoppert, 2016-03-19:
This is the parameter you will want to adjust in QGroundControl:
```c++
/**
 * Vicon position standard deviation.
 *
 * @group Local Position Estimator
 * @unit m
 * @min 0.01
 * @max 1
 * @decimal 3
 */
PARAM_DEFINE_FLOAT(LPE_VIC_P, 0.05f);
```

holmeswk, 2016-03-19:
does this estimate gyro and accel offset and scaling errors?

jgoppert, 2016-03-19:
It does accel bias and gyro bias currently.  It is easy to extend to add scaling errors.

holmeswk, 2016-03-19:
Are you updating the parameters so the sensors module publishes correct data on the sensors combined topic?

Sorry for asking questions that could easily be answered by looking through the code. I've got a working solution and don't have the extra time to do the inevitable trouble shooting that comes with new software.

jgoppert, 2016-03-19:
The sensor combined topic only has the raw data. Biases are not applied. This has been developed over the past year so it is  stable.

jgoppert, 2016-03-19:
Lpe still uses attitude estimator q like inav so nothing attitude related will change.

holmeswk, 2016-03-19:
gotcha. Thanks for responding!

closed 2016-03-19 by holmeswk