PX4/PX4-Autopilot #4031: att_estimator_q rate estimation noise

kd0aij, 2016-03-17:
plot of pitchRate from gyro0,1 and attitude estimator (2016-02-20, FMUv4, git version 0a4d0f6e):
[image]

kd0aij, 2016-03-17:
something is very wrong with a later log (2016-03-11, FMUv4, git version b7a2e5aa):
[image]

jgoppert, 2016-03-18:
Were you running a position estimator at the time? I've seen some bad attitude estimation result from acc comp when my position estimator had a bad parameter.

LorenzMeier, 2016-03-18:
Would be a good idea to disable acceleration compensation for a test. And to test EKF2 in comparison.

kd0aij, 2016-03-18:
After fixing the bugs in sdlog2, I find that the Q attitude estimator appears to predict the future:

[image]

kd0aij, 2016-03-18:
actual flight with more accurate timestamping in sdlog2:

[image]

perhaps there is a feedforward or derivative term in the rate estimation which is a bit too large

kd0aij, 2016-03-18:
hovering with pitchRate near zero:

[image]

LorenzMeier, 2016-03-19:
Does rate control feel better when you switch the Q estimator to just pass on the best voted gyro rate instead?

kd0aij, 2016-03-19:
I haven't tried that. By "switch" you mean modify the code?

LorenzMeier, 2016-03-19:
Yes, in attitude_estimator_q copying the gyro values (and subtracting the offset) instead of using the filter's own estimate.

kd0aij, 2016-03-19:
I'll give that a try. Regarding detailed analysis of the signals, wasn't there a complete rework of sdlog2 underway last summer? I don't trust the sdlog2 timestamps or the plots generated by FlightPlot.

LorenzMeier, 2016-03-19:
The rework went to the back burner due to lots of other ongoing things. Its essentially ready for integration and you would be most welcome to pick it up. We won't be able to pick it up for another 4-8 weeks until I have staffed all open positions.

kd0aij, 2016-03-19:
I'd be willing to work on that; I have a number of things on hold that require good quality logs. Is there a branch in PX4/Firmware?

LorenzMeier, 2016-03-19:
This is ready for testing: https://github.com/PX4/Firmware/pull/2780. The most recent flight plot version should be able to read and graph the resulting file.

kd0aij, 2016-03-19:
@LorenzMeier I tried to rebase https://github.com/PX4/Firmware/pull/2780 on current master, but there were many conflicts. WIll it require a CMake expert solve that problem?

jgoppert, 2016-03-19:
If you want me to take a look, let me know.

kd0aij, 2016-03-20:
@jgoppert I'd appreciate advice on how to get https://github.com/PX4/Firmware/pull/2780 rebased on current master, since I'd like to use the new logging on FMuV4 to test ACRO mode performance

dagar, 2016-03-20:
@kd0aij squashing it together and merging to master wasn't so bad, but something is still broken. I pushed the branch uORB_autogeneration_rebase to the main repo. Do you want to try and get it working?

https://github.com/PX4/Firmware/compare/uORB_autogeneration_rebase?expand=1

LorenzMeier, 2016-05-22:
Fixed.

closed 2016-05-22 by LorenzMeier