PX4/PX4-Autopilot #4038: Mixer tables contain a lot of redundant data

SimonWilks, 2016-03-19:
A while ago I did a couple of experiments to address this issue. I have two branches which freed up several hundred bytes space (IIRC) which allowed even the most complex mixer (Albatross) to be loaded including gimbal mixers and there was still some space free.

The two approaches are based on the knowledge that there is a lot of redundant/wasted information in mixer files. 

The first approach was to simply assume that there are often quite a number of rows that have a default set of values. I changed to using pointers so that a null entry would imply a default mixer entry. Code: https://github.com/PX4/Firmware/compare/master...UAVenture:px4io_memory?expand=1

The second approach used a simple lookup table, again with the assumption that the majority of values will be used multiple times. That way you are mostly storing int8 values which point to the mixer entry in the index (here it's just a fixed size array but it should be easy to know the required size prior to creation). Code: https://github.com/PX4/Firmware/compare/master...UAVenture:px4io_memory_v2?expand=1

I won't be able to finish this off anytime soon, but perhaps it might help someone else finish it off or provide some inspiration.

LorenzMeier, 2016-03-24:
We freed a lot of space on IO thanks to a patch from tridge. It should behave now.

LorenzMeier, 2016-12-28:
Still tend to rather spend the memory without getting fancy on the mixer handling itself, so closing.

closed 2016-12-28 by LorenzMeier