PX4/PX4-Autopilot #4040: Geofence with polygon seem no more supported

MikaelFerland, 2016-03-19:
@dagar Do you think that can we make it work again? Because it pretty useful to build a limit around a mission.

dagar, 2016-03-19:
It's still there, but not exposed in QGC. Have you tried this? https://pixhawk.org/users/geofence

MikaelFerland, 2016-03-19:
Yes, I tried this abd I searched for GF_ON in QGC in parameter list. Do I
have to activate it in the startup file in the px4? Do the path will be
create as well when GF_ON will be set to 1?

dagar, 2016-03-19:
GF_ON was replaced with GF_ACTION. Can you try it again with GF_ACTION set to 1 which will log the geofence violation in the mavlink console.

MikaelFerland, 2016-03-20:
Ok perfect, I will test it tomorrow during flight. Thank you for the update. :smile:

MikaelFerland, 2016-03-21:
Ok, I've tested the geofence, but I got "Critical: Geofence violation for waypoint 2" error when I add a change speed waypoint. This have been done here https://github.com/PX4/Firmware/pull/3567.

dagar, 2016-03-21:
Should be a quick fix to update the mission feasibility checker for NAV_CMD_DO_*.

https://github.com/PX4/Firmware/blob/b323908fc508c400549a8feaae7085c99818b21f/src/modules/navigator/mission_feasibility_checker.cpp#L165

MikaelFerland, 2016-03-21:
You are kind of very effective, I was looking how I could implement that, but you have already done done fix so. Thank you.

commit 408d147a (2016-03-22): MissionFeasibilityChecker geofence ignore NAV_CMD_DO_*

-closes #4040
closed 2016-03-22 by AndreasAntener