PX4/PX4-Autopilot #4054: hmc5883 reports buffer overflow

julianoes, 2016-03-22:
There seems to be an error in the perf reporting of the hmc5883 for the external mag.

```
hmc5883_buffer_overflows: 11315 events
```

@sanderux FYI

LorenzMeier, 2016-03-24:
Not an error. The read queue overflows because we're not using the read() API (we use uORB). If you look at a perf counter you need to read the code related to it. You really don't know what it means else, and its not intended as a system health reporting structure.

closed 2016-03-24 by LorenzMeier

julianoes, 2016-03-24:
Ok, however it is still confusing because an overflow is essentially a red flag thing and gets user (like me) worried. Also, it doesn't shop up for other sensors.

LorenzMeier, 2016-03-24:
Performance counters are not for users. They are instrumentation, not reporting.

julianoes, 2016-03-24:
It's still not consistent with other drivers, even for me as a developer :).

LorenzMeier, 2016-03-24:
You're still confusing it with reporting. Just don't. You can try to make it consistent, but the prime mistake is assuming that it means anything without reading the code.