PX4/PX4-Autopilot #4066: Disarming inflight on CUSTOM_MODE_ENABLED

pickledgator, 2016-03-23:
In commander.cpp, there's a required check for an arming transition and I'm noticing that sending command longs with the CUSTOM_MODE_ENABLED flag is tripping the arming transition to disarm in flight. Specifically, line 673 of commander.cpp: 

```
bool cmd_arm = base_mode & MAV_MODE_FLAG_SAFETY_ARMED;
arming_ret = arm_disarm(cmd_arm, mavlink_fd, "set mode command");
```

Should this check be wrapped in an if statement like the rest of the base_mode checks so that it only executes if a MAV_MODE_FLAG_SAFETY_ARMED flag is sent as param1?

dagar, 2016-11-08:
It's a bitfield which contains the arming state. You can first copy it from the HEARTBEAT before sending MAV_CMD_DO_SET_MODE. There are other messages you can use for initiating takeoff, land, mission, rtl, loiter which don't involve this.

julianoes, 2016-11-08:
Related to: https://github.com/mavlink/mavlink/pull/629

julianoes, 2016-11-08:
Since the mavlink change went in, I should remove this:
https://github.com/PX4/Firmware/blob/master/src/modules/commander/commander.cpp#L721

dagar, 2016-11-08:
Yes! Thank you.

PR #5987 (2016-12-11): commander: ignore arm/disarm in DO_SET_MODE cmd
According to https://github.com/mavlink/mavlink/pull/629 the mavlink
command DO_SET_MODE should only determine the mode but not the
armed/disarmed state, so the MAV_MODE_FLAG_SAFETY_ARMED bit should be
ignored.
Instead the mavlink command COMPONENT_ARM_DISARM should be used instead.

Therefore, the commander now ignores the arm/disarm bit.

Tested in SITL with `commander takeoff` and take off and land using QGC.

Please cross-test, review @dagar.

Fixes #4066.
closed 2016-12-11 by LorenzMeier