PX4/PX4-Autopilot #4076: Add support for Gazebo's <spherical_coordinates> to set initial GPS in sitl_gazebo.

dustingooding, 2016-03-25:
Right now, sitl_gazebo uses a hard-coded GPS coord in Zurich. This prevents the simulation of GPS-waypointed missions in Gazebo that don't occur in Zurich.  Gazebo worlds have a <spherical_coordinates> (http://sdformat.org/spec?ver=1.6&elem=world#world_spherical_coordinates) setting that redefine Gazebo's origin to be a specific GPS coordinate.  If sitl_gazebo used this value (if set) and fell back to a default (like Zurich or Geneva or Lebanon, Kansas), then developers could test missions in simulation prior to running them on-vehicle.  Accurate GPS is more than just nice ground graphics, too, as it can be used for accurate topologies (if they exist).

dustingooding, 2016-03-25:
https://github.com/PX4/sitl_gazebo/pull/18

mhkabir, 2017-04-03:
Done.

closed 2017-04-03 by mhkabir