PX4/PX4-Autopilot #4079: VTOL still increases altitude during front transition

sanderux, 2016-03-25:
http://logs.uaventure.com/view/PJYu2ARidpuQyFsX9qxdKm#Local_Z_PLOT

sanderux, 2016-03-26:
In SITL the climb can be "solved" by setting VT_FW_PITCH_TRIM to -0.25
/cc @AndreasAntener

AndreasAntener, 2016-03-26:
This log shows that the Z velocity setpoints during transitions are negative, -1.2 m/s. That's the reason it climbs. Most likely this has something todo with the vtol takeoff/land items.

sanderux, 2016-03-26:
SITL on master also climbs, if you set transition airspeed to 20 it will climb 20m and then timeout

sanderux, 2016-03-26:
This is from SITL on master: 
Red = Pitch, Green = PitchSP
[image]

tumbili, 2016-05-03:
@sanderux Is this still up to date? Do you experience this problem in real flight?

sanderux, 2016-05-03:
Not on the first prototype of the deltaquad. Maybe on the new one. Fw pos ctl still needs to be enabled. Ada made a start with that

tumbili, 2016-06-22:
Fixed in #4521

closed 2016-06-22 by tumbili