PX4/PX4-Autopilot #4080: sdlog issues

sanderux, 2016-03-25:
- OUT1 logged with OUT0 values
- strange uint8 values

[image]

sanderux, 2016-03-26:
Also, logging in SITL no longer works.
Reverting this commit: https://github.com/PX4/Firmware/commit/2dcd529ad873555a9a67002509f0530c6fb01c76 brought back SITL logging. Not tested for actual flight log problems yet.

@LorenzMeier

bo-rc, 2016-03-26:
#4087 seems to fix the SITL bug.

LorenzMeier, 2016-03-29:
@sanderux It all looks correct code-wise. Are you sure this is happening on master?

sanderux, 2016-03-29:
Might have been related to the double locking. I got a user report i'll check

LorenzMeier, 2016-03-30:
I'm closing this meanwhile. Please re-open if its a real issue on master.

sanderux, 2016-03-30:
At least ion sitl;
- actuators out0 and out1 are identical
- uint8 values remain 0 (at least the ones i tested)

[image]

sanderux, 2016-03-30:
On latest master the uint8 values get random spikes:
[image]

sanderux, 2016-03-30:
@LorenzMeier would be very nice to have this fixed before i go out test flying this weekend

LorenzMeier, 2016-03-30:
Can you share the exact steps in SITL to repro for me?

sanderux, 2016-03-30:
```
git checkout master
make posix_sitl_default gazebo_standard_vtol
```
fly mission
ctrl-c after LANDING DETECTED
open log from ~/src/Firmware/build_posix_sitl_default/src/firmware/posix/rootfs/fs/microsd/log/2016-03-30 in flightplot
add out0 group and out1 group

LorenzMeier, 2016-03-30:
Fixed by #4124.

sanderux, 2016-04-02:
uint8 issues remain
[image]

LorenzMeier, 2016-04-02:
@sanderux Just fixed it. Please check.

commit 0d94800b (2016-04-02): VTOL: Add missing fields to log. Fixes #4080
closed 2016-04-02 by LorenzMeier