PX4/PX4-Autopilot #4083: MC: Won't start after arming (1.2 Stable)

tops4u, 2016-03-26:
Second time yesterday I had the problem, that after pushing the Safety Switch and after Arming by RC Command (both confirmed by the Buzzer) on a AUAV-X2 Board I wanted to take off. But while pushing Throttle up just nothing happened. Several Arm/Disarming did not help. Only remove Power and a complete Reboot let the Kopter take off. Sadly no GroundStation was connected. Take off was attempted in all cases in Stabilized Mode, ESC are properly connected and stopped Beeping the PWM Signal Lost after Boot was complete. PX4 1.2 Stable was running. 

Log Files attached. Maybe it would be an Improvement to Save Messages also in case of a non successful flight, as only the successful Flight after Powercycling was logged with a Messages Log. 

[UnableToFlyLogs.zip](https://github.com/PX4/Firmware/files/190483/UnableToFlyLogs.zip)

LorenzMeier, 2016-03-29:
@tops4u Yes, the log was related to your issue.

LorenzMeier, 2016-03-29:
@tumbili The issue he has is likely related to the land detection state when arming outside of manual. Can you have a look at this with @julianoes ? It looks like he had the same issues yesterday.

LorenzMeier, 2016-03-29:
@tumbili To repro: arm in althold, then try to take off.

tops4u, 2016-03-29:
Just to comment, I'm pretty sure that the FM Switch was in Stab Mode while Arming, since I check all Switches when switching on Transmitter and powering vehicle. Possibly I moved the Vehicle after pushing the safety Switch, but surely before Arming via RC.

xn365, 2016-04-05:
[https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L1274]
When we disarmed in non-manual modes,we are in landed state.Vertical position setpoint always set to zero in landed state,so we cant take off in non-manual modes.

LorenzMeier, 2016-04-05:
@xn365 Thanks! I have a patch I'll send in a sec which should address this.

LorenzMeier, 2016-04-06:
This is all fixed.

closed 2016-04-06 by LorenzMeier