PX4/PX4-Autopilot #4090: BirdsEyeView FireFLY6 VTOL yaws automatically in hover

kPanesar, 2016-03-28:
@tumbili I did a quick hover test with the FireFLY6 on the PX4 flight stack. The FireFLY6 keeps yawing to the right (like spinning vigorously in hover) and I'm not sure why. While examining the [log](http://logs.uaventure.com/view/X7gL7i6frhVpVbezqH9HEK#Actuators_PLOT) I see that while 3 of the motors are spinning at a high rpm, the other three are not. 

I can see all the RC channel values properly in QGC. Also, I have channel 1 mapped to throttle but the log shows channel 3 changing.Not exactly sure if it's a configuration error or a bug.

mhkabir, 2016-03-28:
Can you give us a flight log please? It's likely an estimation issue.

kPanesar, 2016-03-28:
@mhkabir Yes, it's linked in the description.

LorenzMeier, 2016-03-28:
@kPanesar Your log looks like the motor assignment or prop directions are wrong. Did you confirm against the build instructions here?

https://pixhawk.org/platforms/vtol/birdseyeview_firefly

kPanesar, 2016-03-28:
@LorenzMeier Hmm all the motors are spinning in the right direction with the top propellers spinning CCW and the bottom ones spinning CW. I also thought about having the motor ports mixed up somehow but it seems unlikely as I'm able to roll left and right and pitch forwards and backwards fine. Could the motor ports still be incorrect?

LorenzMeier, 2016-03-29:
Sure, anything could be mixed up given your output.

kPanesar, 2016-03-30:
@LorenzMeier Alright, I'll take a look at the motor assignments again and get back to you hopefully this coming Saturday.

kPanesar, 2016-04-03:
Update: I did indeed have the motors mixed up. I was able to do a stable hover with the default parameter values (the ones for the high-efficiency bundle cause oscillations). Thanks for your help, everyone!

closed 2016-04-03 by kPanesar