PX4/PX4-Autopilot #4094: Add mission items to sdlog

sanderux, 2016-03-28:
It would be nice to have the ability to log / extract the mission that was loaded

sanderux, 2016-08-05:
@AndreasAntener are you still panning to implement this?

dagar, 2016-08-05:
You can get a lot of through this GPSP. What else would you like to see?
We should make sure whatever you need is in logger.

sanderux, 2016-08-05:
Mission commands and altitude

dagar, 2016-08-05:
I'm guessing you're aware of GPSP type and altitude, but want to see what the actual mission item values were before going through navigator?

sanderux, 2016-08-05:
yes, but mostly i want to see commands and plot the way the mission was, also if it fails

dagar, 2016-11-08:
How could this fit in with logger? We never publish the mission items.

AndreasAntener, 2016-11-08:
I guess we should add a new topic and log message. And also add a new log message for commands if we're already on ;)

dagar, 2017-04-12:
Is what we have in logger sufficient?

closed 2017-04-12 by dagar