PX4/PX4-Autopilot #4098: MAVLink app: Log messages not making it to QGC, except for severity high

LorenzMeier, 2016-03-28:
@jgoppert This is to track the issue you had with logging. I'll try to repro.

LorenzMeier, 2016-03-28:
@jgoppert This might be a QGC issue.

jgoppert, 2016-03-29:
Thanks for looking into it. I was using 2.9.4.

LorenzMeier, 2016-03-29:
This is working for me on current master with current master QGC.

jgoppert, 2016-03-29:
Yeah, works for me now too, odd. Thanks.

jgoppert, 2016-03-29:
This hasn't gone away, it just works in SITL. I suspect when LPE was modified for the new format, that there is some issue on nuttx with putting the mavlink pub in a class that is created on the stack? Any ideas @LorenzMeier? I'm going to try putting pub declaration in global context and see if that works. 

```c++
__EXPORT void mavlink_vasprintf(int severity, orb_advert_t *mavlink_log_pub, const char *fmt, ...)
{
	// TODO: add compile check for maxlen

	if (!fmt) {
		return;
	}

	if (mavlink_log_pub == NULL) {
		return;
	}

	struct mavlink_log_s log_msg;

	log_msg.severity = severity;

	va_list ap;

	va_start(ap, fmt);

	vsnprintf((char *)log_msg.text, sizeof(log_msg.text), fmt, ap);

	va_end(ap);

	if (*mavlink_log_pub != NULL) {
		orb_publish(ORB_ID(mavlink_log), *mavlink_log_pub, &log_msg);

	} else {
		*mavlink_log_pub = orb_advertise(ORB_ID(mavlink_log), &log_msg);
	}
}
```

jgoppert, 2016-03-29:
@LorenzMeier Yeah, moving it from the class to the lpe cpp file in global context fixed it, just a heads up, might happen elsewhere.

julianoes, 2016-03-30:
@jgoppert: do you have an explanation? I don't understand why this happens. Which platforms does it effect?

LorenzMeier, 2016-03-30:
Without having read the diff I'm pretty sure its because James allocated the publications in a different thread context than where they were used to publish.

closed 2016-03-30 by LorenzMeier