PX4/PX4-Autopilot #4099: Estimate wind direction & land into wind for small wings

LorenzMeier, 2016-03-29:
Discussion: http://discuss.px4.io/t/px4-flying-wing/266/3

tubeme, 2016-04-10:
There is New Zeland autopilot that detects the wind (probably with loiter) and then makes approach for a parachute landing in such a matter that the plane lands in close proximity to the landing mission point. 

Of course it depends on the parachute speed. But if we preset the parachute down speed it is possible to calculate it very precisely geven we have a decent estimation of wind speed and direction.

jcooper21, 2016-04-20:
i would like to add to the diagram that is linked for the landings. The reverse thrust feature with RPM logging for better fixed wing landings in APM which is really good, sensefly also does this in the wind direction autolands.. It would be great to see this in PX4 as well as the parachure failsafe features for emergency landings.

dagar, 2016-06-29:
This is a cool idea. I think I can fit it in with some other fixed wing landing ideas. Then your landing could be either 
 1. in place, into wind
 2. at home, into wind
 3. at predefined landing waypoint following approach (MAV_CMD_DO_LAND_START)

I'll do some testing to see what real wind measurements look like, but does anyone know what would be appropriate for determining landing direction? Loiter for a few revolutions and wait for the wind direction to settle, or just go with an average?

jcooper21, 2016-06-29:
My thought would be loiter for 2-3 revolutions for wind.  Let me know when you are ready t have it tested and I will test it.  I think the most common setting used will be at home into wind and if I am not mistaken we should just have to define the planes glide slope so it knows how far to fly back for the landing setup.

jcooper21, 2016-06-29:
I think ideally you would want this to replace the current "land" function then px4 would have a circle down landing and a regular landing.  The big issue with the current land function is you never know where your plane is going to land so we avoid using it at all costs.

dagar, 2016-06-29:
The best procedure at the moment is to add landing to the end of your mission with a waypoint before the landing point setting up for a nice approach.

jcooper21, 2016-06-29:
That's exactly what I do right now. I think this will make the landing so much better all around for PX4.

dagar, 2016-06-29:
I'm surprised it's so unreliable for you. It does take some tuning to set an appropriate glide slope, etc. As long as I get close to hitting the last approach waypoint I have reliable landings. The problem is the acceptance radius is huge (~50m sphere) so it can start the landing too high, too soon, etc.

jcooper21, 2016-06-29:
The reason it is so unreliable for me is due to wind.  Where I live its not uncommon for wind to quickly change direction and velocity.  It make for very challenging landings.

jcooper21, 2016-06-29:
This will also be very beneficial for the comment in the ground station you just made about not showing the waypoints. Finish a survey mission and the plane goes into this landing mode or circle down landing mode and there's no reason to drop waypoints for landing. The whole surgery setup becomes much easier.

dagar, 2016-07-05:
I'll come back to this once EKF2 is publishing wind. I'm working on a few other FW landing related improvements that I'll open PRs for over the next week. LOITER_TO_ALT, DO_LAND_START, etc.

jcooper21, 2016-07-29:
@dagar It seems that to optimize landings on a fixed wing the global waypoint acceptance radius should not apply in landing conditions since we normally want this to be much more accurate.  Thoughts?

dagar, 2016-07-29:
@jcooper21 lately I've had relatively good results lining up the landing using loiter (alt) with exit at heading alignment and resetting xtrack to the exit location. You'll need recent PX4 and QGC.

jcooper21, 2016-10-02:
@dagar Am I correct in seeing the wind mavlink messages were just implemented in master for EKF2?  That will allow us to test out this feature?

dagar, 2016-10-02:
If wind is now available (again) we can start thinking about how this would be implemented.

dagar, 2017-01-15:
Any opinions on if this should be in place or try to land at HOME, or configurable?

As part of this and RTL -> FW mission landing (#6315) there's a bit of confusion I'd like to try and straighten out.
 - LAND mode (soon with optionally respecting wind)
 - missions with DO_LAND_START
 - RTL landing.

There's also a landing abort command, but it's only respected in mission mode.

closed 2018-02-21 by PX4BuildBot