PX4/PX4-Autopilot #4100: top equivalent for QuRT

LorenzMeier, 2016-03-29:
@jywilson @mcharleb It would be great to have the current CPU load per thread for QuRT, so we can have a top-like output:

```
Processes: 22 total, 3 running, 19 sleepin
CPU usage: 36.81% tasks, 0.76% sched, 62.43% idle
Uptime: 1184.019s total, 779.535s idle

 PID COMMAND                   CPU(ms) CPU(%)  USED/STACK PRIO(BASE) STATE 
   0 Idle Task                  779534 62.433     0/    0   0 (  0)  READY 
   1 hpwork                      48511  4.106   900/ 1592 192 (192)  w:sig 
   2 lpwork                       6706  0.532   572/ 1592  50 ( 50)  w:sig 
   3 init                         1617  0.000  1612/ 2496 100 (100)  w:sem 
 208 top                          2173  3.346  1228/ 1696 100 (100)  RUN   
  85 gps                          2675  0.228   732/ 1192 220 (220)  w:sem 
  87 dataman                        23  0.000   652/ 1192  90 ( 90)  w:sem 
 107 sensors                     34440  2.965  1708/ 1992 250 (250)  w:sem 
 109 commander                    4673  0.380  3156/ 3592 140 (140)  w:sig 
 110 commander_low_prio           2900  0.000  1204/ 2872  50 ( 50)  w:sem 
 113 px4io                       36581  3.041   956/ 1496 240 (240)  w:sem 
 121 mavlink_if0                 11924  0.988  2452/ 2696 100 (100)  w:sig 
 122 mavlink_rcv_if0               665  0.000  1276/ 2096 175 (175)  w:sem 
 132 mavlink_if1                  9833  0.836  2452/ 2696 100 (100)  w:sig 
 147 sdlog2                      14304  1.140  2204/ 3392 177 (177)  w:sem 
 175 attitude_estimator_q        68224  5.779  2196/ 2496 250 (250)  w:sem 
 178 position_estimator_inav     30296  2.585  4844/ 5296 250 (250)  w:sem 
 182 mc_att_control              27820  2.281  1316/ 1496 250 (250)  w:sem 
 187 mc_pos_control               8333  0.684  1180/ 1896 250 (250)  w:sem 
 190 navigator                      47  0.000   844/ 1496 105 (105)  w:sem 
 202 mavlink_if2                 91909  7.908  2452/ 2696 100 (100)  w:sig 
 204 mavlink_rcv_if2               180  0.000  1188/ 2096 175 (175)  w:sem 

Processes: 22 total, 3 running, 19 sleepin
CPU usage: 36.81% tasks, 0.76% sched, 62.43% idle
Uptime: 1184.019s total, 779.535s idle

 PID COMMAND                   CPU(ms) CPU(%)  USED/STACK PRIO(BASE) STATE 
   0 Idle Task                  779534 62.433     0/    0   0 (  0)  READY 
   1 hpwork                      48511  4.106   900/ 1592 192 (192)  w:sig 
   2 lpwork                       6706  0.532   572/ 1592  50 ( 50)  w:sig 
   3 init                         1617  0.000  1612/ 2496 100 (100)  w:sem 
 208 top                          2173  3.346  1228/ 1696 100 (100)  RUN   
  85 gps                          2675  0.228   732/ 1192 220 (220)  w:sem 
  87 dataman                        23  0.000   652/ 1192  90 ( 90)  w:sem 
 107 sensors                     34440  2.965  1708/ 1992 250 (250)  w:sem 
 109 commander                    4673  0.380  3156/ 3592 140 (140)  w:sig 
 110 commander_low_prio           2900  0.000  1204/ 2872  50 ( 50)  w:sem 
 113 px4io                       36581  3.041   956/ 1496 240 (240)  w:sem 
 121 mavlink_if0                 11924  0.988  2452/ 2696 100 (100)  w:sig 
 122 mavlink_rcv_if0               665  0.000  1276/ 2096 175 (175)  w:sem 
 132 mavlink_if1                  9833  0.836  2452/ 2696 100 (100)  w:sig 
 147 sdlog2                      14304  1.140  2204/ 3392 177 (177)  w:sem 
 175 attitude_estimator_q        68224  5.779  2196/ 2496 250 (250)  w:sem 
 178 position_estimator_inav     30296  2.585  4844/ 5296 250 (250)  w:sem 
 182 mc_att_control              27820  2.281  1316/ 1496 250 (250)  w:sem 
 187 mc_pos_control               8333  0.684  1180/ 1896 250 (250)  w:sem 
 190 navigator                      47  0.000   844/ 1496 105 (105)  w:sem 
 202 mavlink_if2                 91909  7.908  2452/ 2696 100 (100)  w:sig 
 204 mavlink_rcv_if2               180  0.000  1188/ 2096 175 (175)  w:sem 
```

Does QuRT have facilities for runtime instrumentation in the scheduler?

mcharleb, 2016-09-20:
@LorenzMeier This is something we also would like to expose. No time frame yet for when it will be done.

LorenzMeier, 2017-06-04:
Closing as stale.

closed 2017-06-04 by LorenzMeier