PX4/PX4-Autopilot #4102: Track down "configuration error" messages

LorenzMeier, 2016-03-29:
@kd0aij Pointers appreciated!

We're trying to find all remaining cases where the system refuses the arm without really telling the user why.

LorenzMeier, 2016-03-29:
User report:
[image]

tumbili, 2016-03-29:
@LorenzMeier Is this in SITL or in real?

LorenzMeier, 2016-03-29:
On a VTOL system by @geoffreybainathsah.

geoffreybainathsah, 2016-03-29:
Hi Guys
to clearify this.
I have a 3DR pixhawk, GPS unit, 3DR power module and digital airspeed sensor
Futaba R7008 runnin on sbus (mode2)
I went through all the sensor steps and managed to get everything green.
After connecting the lipo battery and trying to arm i get the messae "config error"
[image]

tumbili, 2016-04-04:
@sanderux You think you could try to reproduce this?

sanderux, 2016-04-04:
I will try, have not seen it before though. @geoffreybainathsah could you export your params and upload here as zip?

geoffreybainathsah, 2016-04-04:
thanks for lookin at this guys
uploading a ZIP is not supported but i managed to upload it to outlook one drive
<iframe src="https://onedrive.live.com/embed?cid=A3E2D1820187FC2A&resid=A3E2D1820187FC2A%21817&authkey=AMfFMQWr4OXqTEA" width="98" height="120" frameborder="0" scrolling="no"></iframe>

LorenzMeier, 2016-04-18:
Fixed this.

closed 2016-04-18 by LorenzMeier