PX4/PX4-Autopilot #4103: VTOL: Quad Throttle problem.

tubeme, 2016-03-29:
We had about 10 smooth transitions today. After the latest corrections to the transitions it performed very well.

But we still have a problem with the throttle response in quad mode. It is obvious right after back transition.

We fly only Manual for now.

We push the throttle to max at the end of the flight, because the quad started dipping. I see that the ThrottleSP responds to the Throttle RC input. But then when I check the "us" that the pwm OUT is giving the motors... it is waaaay too low on the basis of what the SP is.

[image]

[image]

I see it gave the most 1600us to the ESC of the quad with exception of two spikes. Which is not OK.

[image]

[image]

Here is the Log file:

http://logs.uaventure.com/view/z4iXLcNEGGdCXiDtejPC5R

We have still some oscillations in the pitch of the plane so don't pay attention to this.

Otherwise the VTOL performs waaaay better this time. Thanks to all of you!!!

Truly thank you 

Vasil

tumbili, 2016-03-29:
@tubeme The problem is that you ran into complete saturation. The mixer is allowed to reduce thrust in order to get some authority on pitch/roll axis so I guess this is what happened.
BTW, you can't just look at one pwm output, you need to take the average of all 4 outputs which will give you the actual thrust value.

sanderux, 2016-03-29:
@tubeme i think the cause of your saturation is motor number 3 (front left)
Do you have a thrust stand? if so, measure the thrust of all your motors, i think you will see this motor only provides half the thrust of your other motors.

I say this based on the fact that you told me the CG is correct.

sanderux, 2016-03-29:
[image]

tubeme, 2016-03-31:
@sanderux You were very correct. Thanks again for that. It turned out that ESC #3 was not giving enough throttle. I re-calibrated the ESC's a couple of times with QGC without success #3 was giving lower throttle curve. I could not confirm if the ESC could be calibrated manually with RC because during the try I fried somehow all my 5 ESC's. Anyway we put brand new ESCs and the problem disappeared. And it gave even further stability. So rock solid quad! And another 10 perfect transitions. Next we will do a strong weather transition test with a gusty wind.

We've had a Stall Test crash though so I will make another issue for it.

sanderux, 2016-03-31:
@tubeme glad to have been of help. you will probably need to re-tune your pid's now

closed 2016-03-31 by sanderux

tubeme, 2016-04-01:
@sanderux I did retouch the PIDs yes indeed :)