PX4/PX4-Autopilot #4104: Debrief racer flight

LorenzMeier, 2016-03-29:
@tumbili @kd0aij this is a flight log from Don, which didn't go so well in the end. Do we have a rock-solid conclusion what on the aircraft failed?

http://logs.uaventure.com/view/wHWyexpZ4q235tE3CuZ2hZ

kd0aij, 2016-03-29:
My guess (not sure if it's 100% reconcilable for an "X" frame):

From the log, it looks like you threw one prop blade, then the other and began rolling out of control.
the accelerometers show a large vibration just prior to the rollrate diverging from the rollrate setpoint. Then the rollrate goes very high and the vehicle rolls through about 720 degrees
there's also a spike in motor current at that point, without a corresponding demand for thrust
[image]

LorenzMeier, 2016-03-29:
@DonLakeFlyer Can you confirm visually if it seems like you lost props or motors pre-impact?

DonLakeFlyer, 2016-03-29:
I went out looking for parts to try to figure it out. It went through a tree on the way down, which looks to have knocked all the other props off. I found all the prop parts in the general location of where it landed except for one. So I'd say that's a definite maybe on losing one blade.

closed 2016-04-01 by LorenzMeier