PX4/PX4-Autopilot #4106: Compiling to use with ROS

marianobianchi, 2016-03-29:
When i compile using catkin_make i get this error:

> -- posix-sitl-default
> CMake Error at px4/CMakeLists.txt:172 (include):
>   include could not find load file:
> 
>     configs/posix_sitl_default
> 
> 
> CMake Error at px4/CMakeLists.txt:184 (message):
>   configs/posix_sitl_default must implement px4_os_prebuild_targets
> 
> 
> -- Configuring incomplete, errors occurred!
> See also "/home/sant/catkin_ws/build/CMakeFiles/CMakeOutput.log".
> See also "/home/sant/catkin_ws/build/CMakeFiles/CMakeError.log".
> Invoking "cmake" failed

If i use this command: catkin_make --cmake-args -DCONFIG=ros_sitl_simple, i get this error:

> fatal error: mav_msgs/CommandMotorSpeed.h: No such file or directory
>  #include <mav_msgs/CommandMotorSpeed.h>
>                                         ^
> compilation terminated.
> make[2]: *** [px4/CMakeFiles/mc_mixer.dir/src/platforms/ros/nodes/mc_mixer/mc_mixer.cpp.o] Error 1
> make[1]: *** [px4/CMakeFiles/mc_mixer.dir/all] Error 2
> make: *** [all] Error 2
> Invoking "make -j1 -l1" failed

CommandMotorSpeed is a deprecated message that was removed from mav_comm/mav_msgs. If i edit mav_comm/mav_msgs CMakeLists.txt and add ComandMotorSpeed.msg to the messages list, i can compile without errors using the second command but when i try to use px4 sitl i get this error:

> [ERROR] [1458252337.200967470, 7.340000000]: Client [/gazebo] wants topic /iris_3/command/motor_speed to have datatype/md5sum [mav_msgs/Actuators/25741daf38ed25442e3a66a855ee8d9c], but our version has [mav_msgs/CommandMotorSpeed/061082200f78e4ff571a825255200c0d]. Dropping connection.

I'm starting using ROS so i don't understand if it is a problem in PX4 Firmware library or if it is just a problem of compatibility between different libraries.

LorenzMeier, 2016-03-29:
The ROS build is deprecated. We recommend PX4 Linux / SITL + MAVROS.

marianobianchi, 2016-03-29:
When you say PX4 Linux, do you mean to compile Firmware code using "make posix_sitl_default gazebo" as it is written in http://dev.px4.io/simulation-ros-interface.html? 

If it is so, is there any way to launch the firmware from ROS? I think the answer is running "roslaunch Firmware iris.launch" but it would be excellent to confirm because i'm a little bit lost here with the documentation....

AndreasAntener, 2016-03-30:
@marianobianchi ATM it's the easiest if you do as you suggested, start PX4 SITL separately and then start ROS and connect via UDP.

A direct from ROS startup is WIP and comming soon: https://github.com/PX4/Firmware/commit/27e630e81e55fd666f0144f1439d309b0b753c8d#diff-63945442d0c74f983e47dd6ba99c75e2

closed 2016-03-30 by AndreasAntener

marianobianchi, 2016-03-30:
Thank you @AndreasAntener! Last question: is there a way to set udp address when running "make posix_sitl_default gazebo"? Because i want to test an application using more than one UAV so i will need to launch two instances of PX4 SITL.

AndreasAntener, 2016-03-30:
I'm not currently aware of this, but there were people asking/trying stuff in this direction recently.