PX4/PX4-Autopilot #4122: Control mode switch on transmitter doesn't retry for unallowed transition

jgoppert, 2016-03-30:
Last night we were doing a lot of LPE + lidar + flow testing and found that when the pilot initially started in position control (switch position, but actually in alt control) while the light was blue and then took off and the light turned green (local position was published), the mode would not switch to position control from alt control until the pos-alt control switch was toggled. In the past, the led would flash red indicating an unallowed transition and keep attempting until it succeeded.

LorenzMeier, 2016-03-30:
Which implies you liked the previous approach better?

jgoppert, 2016-03-30:
The existing setup is ok if there were some feedback to the pilot that he was not in position control. The problem is the light is green and the pilot thinks position control is drifting, but he is actually still in alt mode.

dagar, 2017-04-11:
Still needed?

closed 2018-02-21 by PX4BuildBot