PX4/PX4-Autopilot #4126: VTOL: Stall Test falure

tubeme, 2016-03-31:
Hi guys,

We've had a day for Stall Test the FW flight of our VTOL. We chose a dead weather in order to have clearer graphics.

Our PR_P and PR_I are a little stronger, but that is how we like if for this part of testing, so don't pay attention to this. In FW it is tracking well.

So the test is simple. Use cruise throttle and speed in manual. 0.6 throttle ~1580us and speed 14m/s then apply pitch up command gradually. Then lower the throttle bellow the middle again gradually pitch up and hold and finally lower even further the throttle and apply full pitch up command. At the end of this roller coaster the plane stalls. So the Bigaole FC does very well stall prevention techniques during all this three phases of flight. Approaching stall Bigaole starts to decrease the Pitch up input from the RC. Once the airspeed is so low that the plane stalls it starts rolling to one side. Gaining this little airspeed during the stall is enough for the Bigaole to level the plane back then again it rolls to the other side, and again the Bigaole gains control and levels it. In this manner finally we have some sort of crash landing, but in 90% of the cases a without any damage to the VTOL. Also from this situation we recovered very easily just by applying throttle. The Bigaole manages to just level the plane and recover.

I understand that you are developing a quad recovery procedures. But I think that very robust stall recovery mechanism will be very good for the novice user even in Manual Stabilized flight.

Here is the whole experiment

The stall speed is about 8-9 m/s 

[image]

In the second image I offset the airpseed to make it visual over the other graphics.

[image]

I could not understand why the PX4 does allow me to stall the plane? Why does it followed its rate SP in the full Pitch Up command and crashed it. I see it gained speed probably after stall recovery but it was faaar away from stall to allow me at that very moment to give full pitch up command. 

Other than that the VTOL hovers transitions and flies perfect!

LorenzMeier, 2016-03-31:
You flew in attitude mode. There is no stall protection because this is a manual flight mode. If you want stall protection don't fly in attitude. Use altitude control or position control.

The same way as a quad will crash if you apply too little throttle in attitude control mode, a plane will crash here. I also don't think that the flight mode name or type does mismatch expectations (except for the ones you had here).

closed 2016-04-01 by LorenzMeier

tubeme, 2016-04-01:
@LorenzMeier So you are saying that I have to reply to a customer with crash for 1-5k USD they did not use the right mode?

In my opinion Stabalized flight is assisted mode, and that is how it is perceived by the average user, most of the time even by the advanced users. When in Stabalized flight despite the computer does not have control over the throttle it can apply stall techniques to safe the airplane or VTOL in FW flight. We used a 60 US simple stabilization 1x2cm with 3 gyros that does it... In our case we had enabled VT_FW_PERM_STAB = 1 in the PX4 which means the plane was in Stabalized despite it reported Manual in QGC.

Actually in the real aviation the stall warning is probably the single most important warning for the pilots. And in smaller planes it is implemented with mechanical sensor not with airspeed which shows its importance to the pilot. 

In most cases stall maneuvers are made without applying throttle. There is even spin recovery procedures that does not include throttle. So stall can be recovered without throttle just by leveling a plane and giving 15deg slope angle. 

Also I've seen a lot of guys with APM and OSD to have a stall warnings on their displays doing FPV. That was very very important for them when in manual Stabilized flight. It is not a bad idea to have a stall warning in the QGC and also to have it as MAV to the OSD.

Stall prevention and recovery is such a fundamentally important thing for the airplanes it should be implemented in all modes except Acrobatic modes. IE rate modes. Making a stall prevention in Stabalized FW Is just making sure that the plane will level itself during the moments of gaining speed. Trying to give 15degrees slope angle and also preventing my input when it is in conjunction of anti stall dynamics.

So now it turns out that you guys implement and work on Awarenes and Avoidance with super high tech sensors and algorithms and did not have the simple most important algorithm implemented in Stabalized. Of course it is easier the have it in Altitude and Position control with the Throttle but why not extend it to Stabalized.

In the era of Drone crashes bad publicity and regulations the producers of more reliable machines will rule the business. 

We always presume our drones are going to be flown by amatures. This way we guarantee safety!!! Soon there will be nasty regulations about this aircraft. The advanced users make mistakes too. So that is why we rely on computers, to take in the moments we are off for some reason.

I dont understand what you mean by this: 
"I also don't think that the flight mode name or type does mismatch expectations (except for the ones you had here)."

tubeme, 2016-04-01:
Notice that we did not remove the throttle. We just held it between 35-45% which should be far enough for stall recovery. Actually doing good stall recovery will be very good for the gliding abilities of the PX4 and make the indestructible glider controller.

LorenzMeier, 2016-04-01:
Not a single multirotor on the market can be flown in stabilized. They only allow position control. It's your responsibility to not allow customers to get into the wrong flight mode. Let us know if you need support in QGC to disable particular flight modes from the configuration options.

dagar, 2016-04-01:
:+1: for hiding flight modes in QGC. If nothing else I'd prefer to not see acro and rattitude as options when flying a plane.

tubeme, 2016-04-02:
I agree with you on the allowed modes. It would be fantastic if we could arm and take off in Altitude Control in quad mode of the VTOL, otherwise we cannot hide Manual mode. 

We've had a problem before with a preset Acro switch to transition from Hover Stabilized to FW Stabilized. The Acro switch was reversed in both modes so we used VT_FW_PERM_STAB = 1 which is again some sort of problem concerning that it reports in QGC as only Manual in FW flight. May be this is corrected now we never tested again. But we will in two days.

Still it is not a bad idea to have a stall warning in QGC and to the OSD even in Manual FW flight. The usual case I've seen here for FPV plane flying is that they fly them Stabilized and half of the time they are gliding to safe power.

tubeme, 2016-04-02:
APM approach.

http://ardupilot.org/plane/docs/stall-prevention.html

jcooper21, 2016-05-03:
Stall prevention on planes is a must as well as stall recovery. With APM I have still had stalls but that was due to bad airspeed sensors. They tried to solve the bad airspeed issue by adding AHRS wind max parameter for which after X wind resistance the flight controller switches to GPS and disregards the airspeed sensor. The solution they came up with half works I believe the only way to do it properly it to have 3 airspeed sensors and mounting the pitot tubes all side by side to eliminate the bad airspeed readings which lead to major issues in stall prevention.

tubeme, 2016-05-03:
@jcooper21 I think using 3 pitot tubes for prosumer or hobbist product is unpractical because of weight and complications. But I've seen decent results with just a gyro, accel and GPS to other smaller controllers.

jcooper21, 2016-05-03:
I agree with you but it leaves a fundamental question to be answered. Is it PX4 pro as in professional or prosumer. If it's professional then that would be the best way for maximum redundancy. With that being said no one says a hobbiest has to use 3 airspeed sensor however best results would be achieved that was for stall prevention.