PX4/PX4-Autopilot #4129: A confusion on the 'idle_speed' value in mixer_multirotor.cpp

perceptive558, 2016-03-31:
https://github.com/PX4/Firmware/blob/master/src/modules/systemlib/mixer/mixer_multirotor.cpp#L365
outputs[i] = constrain(_idle_speed + (outputs[i] (1.0f - _idle_speed)), _idle_speed, 1.0f);
if we look into this constrain,
_idle_speed < _idle_speed + (outputs[i] (1.0f - _idle_speed)) < 1.0f
Can be writed in this form:
0 < (outputs[i] * (1.0f - _idle_speed)) < (1.0f - _idle_speed)
If _idle_speed not equal to 1.0f, then the former equation can be writed as:
0 < outputs[i] < 1.0f
I do not know whether this result is the one desired

LorenzMeier, 2016-04-01:
@tumbili Please inspect.

tumbili, 2016-04-04:
@perceptive558 you have shown that the original outputs which are calculated are between zero and one. This is ok and the reason why we constrain them at the end.

closed 2016-04-04 by tumbili