PX4/PX4-Autopilot #4138: Clean up warnings/log output about broadcast and mavlink interfaces

julianoes, 2016-04-01:
There are confusing, and too many warnings like
```
WARN  getting broadcast address failed`
```
```
WARN  sending broadcast failed
```

This needs review, and before trying to broadcast, `network_interface_found` should be checked.

Reported by @liamstask .

liamstask, 2016-04-01:
thx! my current setup: on my home network, running sitl_gazebo at f4f1f5f0273ba39af21e5fa98d0513f735005e33.

In `Mavlink::init_udp()` i see `WARN  getting broadcast address failed: Invalid argument` from the attempt to get a broadcast address (I added the `strerror()` detail there locally). Then periodically `WARN  sending broadcast failed, errno: 49: Can't assign requested address` as it attempts to broadcast with an invalid address.

Connecting QGC on localhost works, since it finds it by getting a response to a HEARTBEAT sent in `Mavlink::send_message()` on the default remote address. Commenting out the `sendto()` on the broadcast address allows it to continue working in this way, so seems that we could disable broadcasting if we know that we didn't find a valid broadcast address during init.

julianoes, 2016-04-04:
@liamstask does e832f7dbf5a180f5ab8d1c2e0a65a39231c45441 fix it for you?

liamstask, 2016-04-04:
@julianoes it fixes it, in that it prevents the continued printing, but is still essentially a workaround, since it appears the root problem is #4141.

Ideal fix would be to resolve #4141 directly. But if not, seems like a clearer fix would be:
* change `_broadcast_reported` to something like `_broadcast_address_valid`
* set `_broadcast_address_valid` with the same result that `network_interface_found` currently gets in `Mavlink::init_udp()`

In this case, the broadcast would not be attempted to be sent at all if we know we don't have a valid address. There's already an error/warn message when getting the broadcast addr fails, so that should be enough to indicate that there's an issue with broadcasting.

julianoes, 2016-07-13:
Adressed in various PRs and lastly #5044.

closed 2016-07-13 by julianoes