PX4/PX4-Autopilot #4142: Proposal: FPV Gliding mode

tubeme, 2016-04-02:
Hello guys,

I propose a very practical flight mode. It is the same as Stabalized flight, but with stall warning and protection inbuilt. 

This mode will allow us to fly FPV airplanes for longer hours gliding half of the time to save battery, and have a stall warning in QGC and on OSD and stall protection during gliding with 0% throttle. Also there could be a spin dive recovery as well.

What do you think?

kd0aij, 2016-04-02:
How does it work? "gliding half of the time to save battery" sounds pretty
intelligent to me. Is the pilot involved in deciding when to glide?

dagar, 2016-04-02:
I need to read up on stall detection and recovery.

For now you can get pretty close with any altitude/position control or full auto and telling TECS to only care about airspeed (FW_T_SPDWEIGHT 2) and prevent it from using throttle (FW_THR_MAX 0). Obviously that's not very user friendly.

The airspeed mode I've been working on would also do what you want https://github.com/PX4/Firmware/pull/3861
It's pass through throttle, but the pitch stick is actually commanding airspeed. If tuned properly it feels almost like flying stabilized mode, but you can't stall. Pitch forward and the plane dives for more airspeed, pitch back and it climbs to reduce airspeed.

tubeme, 2016-04-03:
@kd0aij Yes it is manual throttle like in Manual-Stabalized. It is just that Manual-Stabalized does not have any stall detection and recovery techniques because probably PX4 does not have control over the throttle. 

The guys here at my green are flying in a mode with their APM that allows them to cut the throttle (ie. manual throttle) but at the same time they have stall detection, prevention or recovery implemented in APM. They receive the stall warning on their OSD. For FPV they are using larger gliders so half of the time they just stop the motor and use air thermals and currents to hold in the air. With their stall detection and protection there is no problem doing this. It just decrease their Roll or pitch input to a safe margins depending on the speed. We've seen a stall recovery in Bigaole FC and on ET Vector. The strategy Bigaole uses is to minimize your roll to a minimum and puthing your nose in a dive position, given the motor is off, not allowing you to pitch up. Once the stall accurs ussually the plane starts spin dives, but Bigaole manages to level the plane in the moments it gains speed.

Even if there is stall warning only it is better than no info on stall. I never seen PX4 recover from spin or from stall. PX4 relies only on throttle for stall recovery only in Alt Pos Control.

Here is how it is implemented in ArduPuilot:

http://ardupilot.org/plane/docs/stall-prevention.html

You can notice that in ArduPilot it is implemented in the roll modes not only to the assisted modes:

> when in roll controlled modes the autopilot will monitor your demanded bank angle and airspeed and work out if you have sufficient margin above the stall speed to turn at the demanded bank angle. If you don’t then the turn will be limited to the safe limit, but it will always allow a bank of at least 25 degrees (to ensure you can still manoeuvre if your airspeed estimate is badly off).

tubeme, 2016-04-03:
@dagar Yes it is similar to your Airspeed, but we want to be able to shut the motor off and glide safely. Technically every plane is able to glide, given a certain slope angle. The difference is the rate of descend for each plane. So in practice we can control the plane to the very ground gliding if there is good level flight. The stall prevention could be the only safe solution if the motor of the airplane stops in emergency and you continue gliding.

tubeme, 2016-04-04:
This mode can be called just **FPV mode** and could be implemented in another way probably even better. It is just like Altitude Control from 5% throttle to 100% throttle - min max airspeed and altitude priority. And once you shut the throttle at 0% it becomes a gliding plane with stall prevention priority. Once you want to get some altitude or move in a direction you just give some throttle and the plane just picks up.

dagar, 2016-04-05:
Constraining roll based on airspeed is a relatively easy change that would apply to any mode other than manual.
FW_T_RLL2THR is pretty close to the 2nd part of APM stall prevention. It might be more intuitive if the TECS min airspeed was just increased instead.

dagar, 2017-04-11:
Still wanted? I could resurrect and test airspeed mode if anyone's interested.

tubeme, 2017-04-11:
Still wanted! I think it is very good mode for the hobbyists. It should have the following properties:

1. It should detect stalls, and once detecting stall it should give proper control commands + Throttle to get airspeed enough to prevent the stall ie it should make sure the airspeed does not fall bellow min airspeed (probably like your airspeed mode). We could provide airspeed by pitching down at e certain angle if no throttle applied.

2. It should shut the engine once bellow 5% throttle.

3. Once throttle > 5% the throttle should be like in assisted modes ie giving different airspeed.

4. It is not necessary to hold altitude because the usual scenario for flying FPV is gaining altitude while going to a certain point, and then gliding back on no throttle or very low throttle 20-30% and pitching down towards the RTL point. That is seen here at our field and also that is the practice I see in most of youtube videos of long fights. Whoever wants to fly holding altitude he could just use Altitude mode.

dagar, 2017-04-11:
Airspeed will automatically pitch down to gain airspeed. It won't try to hold altitude. If I resurrect this are you interested in testing?

tubeme, 2017-04-11:
Yes I will test it! Making just the right plane for this kind of tests.

closed 2018-02-21 by PX4BuildBot