PX4/PX4-Autopilot #4145: Strange takeoff behavior on mission

sanderux, 2016-04-02:
When executing a just uploaded mission, sometimes, the takeoff tries to move laterally towards a waypoint. I have seen this behavior in SITL too. Rebooting solves the problem.
I think it happens when having armed it and then upload a new mission

I will try to find a log and also try to reproduce.

tumbili, 2016-04-04:
@sanderux Let me know if you were able to repro it, curious to see the log.

tumbili, 2016-04-11:
@sanderux Ok, I was also able to repro this.

sanderux, 2016-04-11:
@tumbili any idea? could it be the new mission reset code?

tumbili, 2016-04-11:
@sanderux I don't know yet, not able to simulate anymore now. Terminal tells me that the socket bind for QGC failed.

tumbili, 2016-04-11:
Do you have a log with you?
Can be a SITL log.

sanderux, 2016-04-11:
No i dont, i have a real life log on the deltaquad

sanderux, 2016-04-11:
can search for it tonight

sanderux, 2016-07-13:
Havent seen this behavior anymore

closed 2016-07-13 by sanderux