PX4/PX4-Autopilot #4149: Check that params are loaded via 433MHz telem

tumbili, 2016-04-04:
Based on user report
http://discuss.px4.io/t/i-used-radio-telemetry-dont-get-parameter/297

tubeme, 2016-04-04:
Two hours ago I flashed master and it loaded the parameters in QGC properly with 433Mhz Telemetry...

The problem can be the telemetry settings. Usually if there is not enough speed due to bad reception or interference it might load the parameters 10 minutes so it appears like it does not load the parameters. It connects in the beginning but then there is not enough bandwidth and it refuses to load the params. May be QGC should report this condition to the pilot... If the case is interference he can lower the speed of the telemetry and it might help.

tumbili, 2016-04-05:
@tubeme Thanks for the feedback!

closed 2016-04-05 by tumbili