PX4/PX4-Autopilot #4151: GPIO issues on PX4FMUv1

mantelt, 2016-04-04:
Hi

I'm trying to use GPIOs on the old PX4FMUv1 (the ones on the 15 pin connector designated "MULTI" on the schematics, GPIO_EXT1 / GPIO_EXT2). I had this working on a quite old version of the firmware  (not yet using cmake), but I recently switched to the newest stable version. Ever since, I'm not able to use those GPIOs anymore.

I'm starting the fmu app in mode_serial, which worked before.

Is there anybody using GPIOs on the old FMUv1 but with a recent FW version?

Also the gpio_led app in the default fw does not work on a FMUv1, while it works on the FMUv2.

Any hints are very much appreciated!

kd0aij, 2016-04-04:
I'd look for changes in src/drivers/px4fmu/px4fmu.cpp
It has changed quite a bit since the addition of fmu-v4.

dagar, 2017-04-11:
Was this ever resolved?

closed 2017-04-12 by dagar

mantelt, 2017-04-12:
We eventually switched to Pixhawk (FMUv2), so I can't tell if it was resolved or not.