PX4/PX4-Autopilot #4157: Add navigator to Snapdragon

julianoes, 2016-04-04:
The navigator module is not yet running on the Snapdragon. I need to add that.

LorenzMeier, 2016-04-09:
@julianoes Is there any particular challenge with this?

julianoes, 2016-04-11:
@LorenzMeier It depends on the dataman which doesn't compile for QURT just yet. And then we'll have to verify the file gets created, etc.

jywilson, 2016-05-03:
@julianoes Is there something you need for dataman, in the way of file I/O functions that we currently don't support?

julianoes, 2016-05-04:
@jywilson I don't think so. It should be straightforward. I'll give it a try soon.

julianoes, 2016-05-13:
I tried this and it turns out the navigator needs dataman which in turn uses `lseek()` which doesn't seem to be available on the DSP side.

For now, I'm going to propse to run the navigator on the Linux side.

LorenzMeier, 2016-05-13:
That seems perfectly fine.

closed 2016-05-22 by LorenzMeier