PX4/PX4-Autopilot #4158: Mag calibration fails over WiFi connection

DonLakeFlyer, 2016-04-04:
This is using a PixRacer over WiFi. Important messages which trigger the QGC state machine to move to the next step get lost, which then causes QGC to get confused. The one which I have noticed is the "side done" message, although I'd assume others could get lost as well. For QGC it's important that it receives the start/done/fail and orientation detected/side done messages. If some of these are missed the ui state machine doesn't work. It seems to be caused by progress messages being output at too high a rate. If I comment out the progress updates mag cal will suceed. Leave them it, and I have 100% failure rate.

julianoes, 2016-04-05:
#4164 should fix the issue. @DonLakeFlyer can you please cross test with your Pixracer? I don't have one available with wifi.

sn0west, 2016-04-05:
Question to you both. This bug has been reported 4 days ago but the issue is only one day old. 

Shall we do a issue immediately or first post it on the forum? 

http://discuss.px4.io/t/v1-3-0-beta-testing/283/9?u=sn0west

What about the other issues in this post?

sn0west, 2016-04-05:
FYI: it still dosn't work with the latest master over wifi.

LorenzMeier, 2016-04-06:
Merged to master only now, so closing as cross-tests were successful.

closed 2016-04-06 by LorenzMeier

sn0west, 2016-04-06:
It still dosen't work with wifi. I did a terminal LOG, where shall I upload this file?

[160406_Compass_Calibration_Problem_LOG_1.txt](https://github.com/PX4/Firmware/files/207240/160406_Compass_Calibration_Problem_LOG_1.txt)

I try to describe what happend. Starting Calibrating the compass over wifi all boxes are red, the first one get yellow, I turn around with my copter, the connection is lost for some seconds, then it comes back I turn the copter on a other side the I get two yellow boxes and the I can finish the calibration.

DonLakeFlyer, 2016-04-06:
It's still missing messages.

julianoes, 2016-04-07:
@DonLakeFlyer that's only for Pixracer on Wifi but not for anything else, correct?

DonLakeFlyer, 2016-04-07:
I can repro with PixRacer on WiFi. Not sure if it's just a WiFi problem of the PixRacer has anything to do with it. USB seems to be ok.

DonLakeFlyer, 2016-04-07:
We're digging into it on the QGC side now to see if we are losing UDP packets.