PX4/PX4-Autopilot #4167: Restricted area funcion

liang-tang, 2016-04-05:
One of the important functions of PX4 controlled machine is Flight restriction, we can't let vehicle flight in restricted area,it's Forbidden and unsafe.So, can we add restricted area function for PX4?

liang-tang, 2016-04-05:
@LorenzMeier

TSC21, 2016-04-05:
https://pixhawk.ethz.ch/mavlink/#SAFETY_SET_ALLOWED_AREA

TSC21, 2016-04-05:
There's a plugin on (mavros)[] to set it: https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/safety_area.cpp

But I don't think this is implemented on Firmware yet. In any case, this is geofencing related.

liang-tang, 2016-04-05:
yes, we can add airport coordinates(to ROMFS?) to set flight restriction area.

dagar, 2016-04-05:
The current geofence system could be extended to do this. How would you want to specify the forbidden area? How many zones would you need to define?

liang-tang, 2016-04-06:
657 forbidden area(large/medium airport ,15913 bytes) i collected, and store in ROMFS,

TSC21, 2017-04-06:
So basically https://github.com/mavlink/mavros/pull/689 defines allowed areas, as a geofence. Probably we can consider that everything out of that area is a no-flight zone @liang-tang? Or maybe we can define a new Mavlink message to restricted flight areas - could be synced automatically to the online restricted UAV flight areas data.

LorenzMeier, 2017-08-20:
I'm closing this since we have a beta airman integration as PR up now.

closed 2017-08-20 by LorenzMeier