PX4/PX4-Autopilot #4169: Can't compile for gazebo on Mac

julianoes, 2016-04-05:
I tried to follow the steps in the devguide to get the gazebo sim working on Mac OS X10.11.4.

Unfortunately `make posiz gazebo` fails to compile as follows:

```
Scanning dependencies of target rotors_gazebo_motor_model
In file included from /Users/julianoes/src/Firmware/Tools/sitl_gazebo/src/gazebo_opticalFlow_plugin.cpp:24:
In file included from /Users/julianoes/src/Firmware/Tools/sitl_gazebo/include/gazebo_opticalFlow_plugin.h:26:
In file included from /usr/local/Cellar/gazebo6/6.5.1/include/gazebo-6.5/gazebo/rendering/Camera.hh:44:
In file included from /usr/local/Cellar/gazebo6/6.5.1/include/gazebo-6.5/gazebo/rendering/ogre_gazebo.h:61:
/usr/local/include/OGRE/Terrain/OgreTerrainPaging.h:33:10: fatal error: 
      'OgrePagedWorldSection.h' file not found
#include "OgrePagedWorldSection.h"
```

LorenzMeier, 2016-04-05:
Will look into this.

nicolaerosia, 2016-04-05:
Can you try https://github.com/PX4/sitl_gazebo/pull/13 ?

LorenzMeier, 2016-04-06:
Fixed, thanks!

closed 2016-04-06 by LorenzMeier