PX4/PX4-Autopilot #4179: build firmware error

molele2, 2016-04-07:
I try builded for firmware, output this error：

 make px4fmu-v2_default
-- nuttx-px4fmu-v2-default
-- could not find gdbtui
CMake Error at cmake/toolchains/Toolchain-arm-none-eabi.cmake:62 (message):
  could not find genromfs
Call Stack (most recent call first):
  /usr/local/share/cmake-3.5/Modules/CMakeDetermineSystem.cmake:98 (include)
  CMakeLists.txt:204 (project)

CMake Error: CMAKE_ASM_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!
ninja -C /root/src/Firmware/build_px4fmu-v2_default  
ninja: Entering directory `/root/src/Firmware/build_px4fmu-v2_default'
ninja: fatal: chdir to '/root/src/Firmware/build_px4fmu-v2_default' - No such file or directory
make: *** [px4fmu-v2_default] Error 1

molele2, 2016-04-07:
this building on centos7

mhkabir, 2016-04-07:
Please install the required packages. Your build system is not setup properly.  
Follow the Devguide. You might need the boutique installation instructions.

molele2, 2016-04-08:
yum install epel-release-7-5.noarch.rpm
yum install python-setuptools
easy_install pexpect
easy_install pyserial
yum groupinstall "Development Tools"
yum install glibc.i686 ncurses-libs.i686
yum install openocd libftdi-devel libftdi-python python-argparse flex bison-devel ncurses-devel ncurses-libs autoconf texinfo libtool zlib-devel cmake

I do not known which lost lib?

molele2, 2016-04-08:
I realy believe this，any boy help me for build error
so I try again
 this error need install python-empy and make this project genromfs-0.5.2
build  again  

ntrol_main.cpp.obj
[ 96%] Linking CXX static library libuavcan_stm32_driver.a
[ 96%] Built target uavcan_stm32_driver
[ 97%] Building C object src/modules/uavcan/CMakeFiles/modules__uavcan.dir/uavcan_params.c.obj
[ 97%] Building CXX object src/modules/uavcan/CMakeFiles/modules__uavcan.dir/actuators/esc.cpp.obj
[ 98%] Linking CXX static library libmodules__fw_att_control.a
[ 98%] Built target modules__fw_att_control
[ 98%] Building CXX object src/modules/uavcan/CMakeFiles/modules__uavcan.dir/actuators/hardpoint.cpp.obj
[ 98%] Building CXX object src/modules/uavcan/CMakeFiles/modules__uavcan.dir/sensors/sensor_bridge.cpp.obj
[100%] Building CXX object src/modules/uavcan/CMakeFiles/modules__uavcan.dir/sensors/gnss.cpp.obj
[100%] Building CXX object src/modules/uavcan/CMakeFiles/modules__uavcan.dir/sensors/mag.cpp.obj
[100%] Building CXX object src/modules/uavcan/CMakeFiles/modules__uavcan.dir/sensors/baro.cpp.obj
[100%] Linking CXX static library libmodules__uavcan.a
[100%] Built target modules__uavcan
Scanning dependencies of target firmware_nuttx
[100%] Building C object src/firmware/nuttx/CMakeFiles/firmware_nuttx.dir/builtin_commands.c.obj
[100%] Linking CXX executable firmware_nuttx
[100%] Built target firmware_nuttx
Scanning dependencies of target build_firmware_px4fmu-v2
[100%] Generating nuttx-px4fmu-v2-default.px4
[100%] Built target build_firmware_px4fmu-v2

closed 2016-04-08 by molele2

Jai-GAY, 2018-03-21:
Today i encountered this error also. found this past issue checked that i had python-empy already installed. Hope to get some pointers to my issue.

CMake Error at cmake/toolchains/Toolchain-arm-none-eabi.cmake:62 (message):
  could not find genromfs

[root@localhost ]# yum info python-empy
Loaded plugins: fastestmirror, langpacks
Loading mirror speeds from cached hostfile
 * base: mirror.vodien.com
 * epel: ftp.kddilabs.jp
 * extras: mirror.vodien.com
 * updates: mirror.vodien.com
Installed Packages
Name        : python-empy
Arch        : noarch
Version     : 3.3.2
Release     : 2.el7
Size        : 497 k
Repo        : installed
From repo   : epel
Summary     : A powerful and robust template system for Python
URL         : http://www.alcyone.com/software/empy/
License     : LGPLv2+
Description : EmPy is a system for embedding Python expressions and statements
            : in template text; it takes an EmPy source file, processes it, and
            : produces output.