PX4/PX4-Autopilot #4180: IC engine throttle problems. Successful flights.

tubeme, 2016-04-07:
Hello guys,

I've had two days of successful PX4 flights with an IC engine plane. We did about an 3 hours of flying missions without a problem. 

But there are some things that does not work for the IC engine setup so you might tweak them a little on order to be OK.

1. We use servo for the throttle! This is very important. Because the kill switch you provided, probably working for electric motors, and MCs, does not work properly for the IC engine. When I flip the kill switch, QGC reports this and the servo just freezes at whatever position is at the moment. For the proper kill switch we need a 950 or 1000us to be fed to the throttle servo once the kill switch is on. This guarantees the servo will cut the gasoline and the engine will stop.

2. If at 950us is our throttle cut then our idle throttle is at 35%. We set FW_THR_IDLE=0.35 and FW_THR_MIN = 0.4. But in reality we don't have the 35% idle throttle all the time when armed. May be this idle is for the auto modes. If this is the case then the implementation of idle throttle is wrong. We need this idle throttle the moment we arm in all modes Acro, Manual, Stabilized... all modes no exceptions, because otherwise the engine could not be started, or if we start it in auto mode and switch to manual mode without this idle throttle applied, the engine will stop. 

We do our own idle throttle now and kill switch with the RC, given the FW_THR_MIN at 0.4 the plane just executed its missions perfect. We gave some slew to the throttle as well in order to prevent choking the engine.

We kicked ArduPlane's ass in the missions. :) They had 2 failures, we performed 4/4.

PX4 Rules!

tumbili, 2016-04-08:
@tubeme Thanks for the feedback.

LorenzMeier, 2016-05-22:
Could you describe what we should change to make it easier for you?

tubeme, 2016-05-22:
For IC Engine Airplanes and Helicopters the procedure is the same. For throttle we have connected a Servo to CH3, which in its term mechanically is driving the gasoline throttle of the IC Engine.

In its zero position the Servo is shutting the gas to the engine. If we accept this position as 1000us or PWM_MIN. In order for the engine to work and have a good flow of gasoline the servo should be at 35% in our case and no lower. So our Zero throttle in the RC becomes 35% offset to CH3 with a switch that arms the IC engine throttle.

We have a three step arming process. 

1. Arm with Button
2. ARM with RC command
3. Switch ON the arming button of the IC engine.

Actually the IC Engine Arming switch becomes a kill switch. Once the Arming switch is put to OFF, it sends the Servo a 1000us, and the servo cuts the gasoline and the engine stops. 

We prefer to have this arming procedure, because the pilot remembers always where the kill switch is.

So in the current implementation of PX4 we have Kill switsh.

In our case it should work in reverse logic. and nothing more. And make sure once the kill switch is ON, it sends predetermined amount of throttle + offset to the throttle Servo. For different engines it is different % throttle that is the minimum throttle. So it should be a parameter that can be changed.

This offset should apply to ALL MODES no exception.

tubeme, 2016-05-23:
Forgot to mention about the starter.

We need another RC switch for the starter if applicable.

So in our taranis we program SG to be IC engine arming/kill switch and SH for the starter.

dagar, 2017-04-12:
@tubeme do you still need this?

tubeme, 2017-04-12:
Not urgently but it is real scenario. This is applicable to Glow engines BTW. Because the gasoline engines cannot be cut with just throttle cut. We use electronic throttle cut for the gasoline engines. I think there should be some thought on combining the two. In one case for throttle cut we use electronics on some channel, and in the other case we use the throttle cut the style I explained here. It is related to #6916.

tubeme, 2017-04-12:
Probably with a couple of parameters and the help of mixer editor we could manage combining the two scenarios - Glow engines and gasoline engines. We put both types to planes and helicopters despite helicopters mostly use glow engines. We have prototype helicopter with boxer gasoline engine.

closed 2018-02-21 by PX4BuildBot