PX4/PX4-Autopilot #4188: Questions about external heading implementation in attitude_estimator

chengshy, 2016-04-08:
I am reading attitude_estimator_q code recently and want to understand the implementation of the filter. I am confused about this external heading support implementation https://github.com/PX4/Firmware/blob/master/src/modules/attitude_estimator_q/attitude_estimator_q_main.cpp#L485-L491, especially ```v(1.0f, 0.0f, 0.4f)``` part. 

I checked the commit history and found this part was originally added in attitude_estimator_ekf. However, I didn't find more documents about this implementation. Could anyone explain this to me or show me the paper I should look into?

Thanks

tumbili, 2016-04-11:
@chengshy I assume the vector 'v' is some kind of magnetic field vector. I agree that it's not well documented.
@mhkabir Could you help him out here?

mhkabir, 2016-04-11:
That's actually not from me. @LorenzMeier introduced that originally in the
old attitude EKF. I just adapted it to attitude_estimator_q.

He would be better able to answer your question.

chengshy, 2016-04-12:
@tumbili @mhkabir Thanks for help. So we are using the external heading for roll pitch estimation as well? @LorenzMeier Would you like to help me out here?

pickledgator, 2016-05-20:
I was playing around with this today and was also having trouble understanding the implementation. Namely, since when vision is enabled, the heading doesn't seem to converge to the correct value. Eg, when sending a 1,0,0,0 quaternion, the attitude converges to a non-zero yaw angle.

mhkabir, 2016-05-20:
Does it change at all when you send the unit quaternion?

pickledgator, 2016-05-20:
Looks like my external vision quaternion was in the wrong reference frame (rotated about X by 90 deg), so despite that a direct mapping from quat->rpy produces the correct yaw value even when roll=90 deg, it looks like that breaks the currently implemented yaw projection/extraction logic.

mhkabir, 2016-05-20:
I heard reports about a singularity in the yaw projection DCM, but it
wasn't confirmed. Likely because the angle it occurred at was rather
difficult to achieve in flight.

Would you mind having a look and seeing if you can find anything bad with
the math?

tuofeichen, 2016-07-28:
hey guys,
I'm still confused about the magic number. Any idea what's it about ?

`math::Vector<3> v(1.0f, 0.0f, 0.4f); `

LorenzMeier, 2016-12-26:
Closing as stale.

closed 2016-12-26 by LorenzMeier