PX4/PX4-Autopilot #4190: PID XY Tuning: quadcopter rotates around the desired position

aitjellal, 2016-04-08:
Hello,
I am using PIxhawk with a motion capture to fly indoor.
My settings:
Frame: 380 cm, 3 Cell Battery, mavros, mocap, ,Qgroundcontrol.
[image]

My attitude controller and "Z" controller seem to work fine. But my "XY" controller is not well tuned.

When I  try to fly in "Offboard mode" to a desired position  the quadcopter keeps rotating around the desired position,  Radus: 0.3m to 0,7m.

In QGC I see there are Velocity and a Position: Controllers (P,  and PID)

- did someone experienced the same behaviour? 
- Which controller should I tune first (Velocity or Position)?
-Which parameters should I change first ?  Should I decrease or increase ?

I attached a screenshot of the current parameters.
I would appreciate if someone can help me.

mhkabir, 2016-04-09:
First off, you're using an ancient version of QGC. What version of the flightstack are you running? 

Secondly, parameters won't help. A log would. Likely that you need to tune your yaw controller, or you have magnetometer interference.

aitjellal, 2016-04-09:
Thank you for the quick reply @mhkabir 
I use QGC just to change the parameters, I think I do not need the newest version.
I am using the newest stable release :  v1.2.0 Release,  installed from file https://github.com/PX4/Firmware/releases 
I am flying indoor in small lab  with Optitrack. 

magnetometer ? I thought  it is only important for GPS navigation. There is no GPS mounted on my Quadcopter. I remap the "Mocap pose" to "mavros/vision_pose"  and check if the "mavros/local_pose" is ok. And I did not check the magnetometer output .

My yaw controller seems to be well tuned when I set a desired yaw, The Quadcopter  rotates and reach. This happen very quicky (aggressive) and does not oscillate.
  
Would this be a problem of the "FeedForward" parameters? 
Thanks.

LorenzMeier, 2016-04-09:
@aitjellal The problem is that your yaw angle and your X/Y position do not agree. Are you using compass heading instead of the mocap heading?

aitjellal, 2016-04-11:
@LorenzMeier  @mhkabir   Yes, I was using the compass Heading (default ATT_EXT_HDG_M =0).
I then changed it to ATT_EXT_HDG_M =1  (for vision because I have remapped "Optitrack pose" to "vision pose")   and still have problems.  The Quadcopter does not fly to the desired position. 

Should I use  "Compass Heading" or "Motion capture heading" ?    Which one is better? 
Should I align the Pixhawk  with the Optitrack before I power-on ? (Make the arrow on pixhawk point to Optitrack +X) ?
How should I proceed to set the Heading correctly?  
Thank you

locustcox, 2016-04-12:
Toilet-bowling is typically caused by an offset in the heading. This error is amplified the closer the multicopter is to its target. I'd highly recommend using mocap heading when available. It's about as close to an exact measurement of the mutlicopter's attitude/heading/pose as you can get. 

So yes, I'd set ATT_EXT_HDG_M to 2. I'd also play around with the weightings. I'd definitely move ATT_W_EXT_HDG higher than its default of 0.1, since it can be relied on much more highly than the compass in an indoor environment. The exact weightings will be for you to determine. Also, make sure to properly align your mocap system and zero out any yaw that appears in vicon (or whatever system you're using) when the multicopter is placed at "zero" yaw in the room.

aitjellal, 2016-04-13:
Thank you @locustcox .
I set ATT_EXT_HDG_M = 1; and ATT_W_EXT_HDG=95
and "tried" to align the pixhawk with the Tracking system. 
For yaw= 0 
I expect to get in the mavros/local_position/pose"
Orientation quaternion=
    x: 0.0
    y: 0.0
    z: 0.0
    w: 1.0

But I get 
  orientation: 
    x: 0.00305893994527
    y: 0.00357438427461
    z: -0.999928040518
    w: 0.0110354179502
(x =0 , y=0,  **z= -1**, w=0.01) 

is this normal ?

thanks

aitjellal, 2016-04-14:
I think there is a bug somewhere in (Mavros ,Mavlink, Px4)
when I align the my Optitrack coordinate frame with Pixhawk 
and use external heading (no compass heading, ATT_W_EXT_HDG=1.0)
The positionn x,y,z  is transmitted corretly
but the rotation (Quaternion x,y,z,w) is not sent correctly.
so when I compare 
mavros/local_position/pose
and 
vrpn_client/pose_world

the positions argees but the rotation do not agree ( , ATT_W_EXT_HDG=1.0)

when I use only the compass  heading , it works.
is this an issue ? is this issue known ?
I think I can  close this issue.
best regards

closed 2016-05-03 by tumbili

NCharabaruk, 2016-11-11:
Was this ever looked into further? I am using a PX4 on a hexacopter. I'm generating a pose estimate using hector mapping and the elevation comes for a sonar sensor. I republish this pose to /mavros/vision_pose/pose. When I look at /mavros/local_position/pose the xyz are right, but the rotation is wrong. For some reason the local position is yawed by 180deg from what I tell it. As well, it drifts back and forth centered on the incorrect yaw even when the drone is still on the ground.