PX4/PX4-Autopilot #4191: INAV / Optical flow sim flaky on master

LorenzMeier, 2016-04-09:
the optical flow sim target for INAV is flaky on master. We should probably rather focus on optical flow validation for EKF2 instead of fixing this.

```
make posix gazebo_iris_opt_flow
```

error:

```
WARN  INAV poll timeout
WARN  Q POLL TIMEOUT
```

LorenzMeier, 2016-04-09:
Same happens if I enable EKF2 instead of INAV and the Q estimator. So I think its rather related to the sim environment / Gazebo and needs debugging on that end.

LorenzMeier, 2016-04-18:
@ChristophTobler How is it working for you?

ChristophTobler, 2016-04-18:
@LorenzMeier I didn't notice anything the last time I used it... Will try again tomorrow

ChristophTobler, 2016-04-19:
@LorenzMeier I just pulled from master and I could not reproduce the behaviour. However, yaw is not working properly (very slow) in the gazebo simulation. It works fine in jmavsim. I have not updated gazebo (version 6.5.1) and it worked before I pulled master.

LorenzMeier, 2016-12-28:
Stale.

closed 2016-12-28 by LorenzMeier