PX4/PX4-Autopilot #4195: Gazebo SITL doesn't find Eigen anymore on Mac

AndreasAntener, 2016-04-09:
@LorenzMeier  @nicolaerosia after the Eigen dependency change to `find_package` it doesn't work anymore for me on mac, not sure why. Eigen is installed via brew.

```
CMake Error at CMakeLists.txt:13 (find_package):
  By not providing "FindEigen.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Eigen", but
  CMake did not find one.

  Could not find a package configuration file provided by "Eigen" with any of
  the following names:

    EigenConfig.cmake
    eigen-config.cmake

  Add the installation prefix of "Eigen" to CMAKE_PREFIX_PATH or set
  "Eigen_DIR" to a directory containing one of the above files.  If "Eigen"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
```

AndreasAntener, 2016-04-09:
Also the optical flow and sensors plugin won't compile:
```
      'gazebo::sensors::RaySensor'; did you mean 'GetLaserShape'?
  this->parentSensor->LaserShape()->DisconnectNewLaserScans(
                      ^~~~~~~~~~
                      GetLaserShape
/usr/local/Cellar/gazebo7/7.0.0-20150824-1969aea/include/gazebo-7.0/gazebo/sensors/RaySensor.hh:189:41: note: 'GetLaserShape' declared
      here
      public: physics::MultiRayShapePtr GetLaserShape() const
                                        ^
/Users/ada/UAV/src/Firmware_cmake/Tools/sitl_gazebo/src/gazebo_lidar_plugin.cpp:69:5: error: no matching function for call to
      'dynamic_pointer_cast'
    std::dynamic_pointer_cast<sensors::RaySensor>(_parent);
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/memory:4927:1: note: candidate
      template ignored: could not match 'std::__1::shared_ptr' against 'boost::shared_ptr'
dynamic_pointer_cast(const shared_ptr<_Up>& __r) _NOEXCEPT
^
/Users/ada/UAV/src/Firmware_cmake/Tools/sitl_gazebo/src/gazebo_lidar_plugin.cpp:78:56: error: no member named 'WorldName' in
      'gazebo::sensors::RaySensor'; did you mean 'GetWorldName'?
  this->world = physics::get_world(this->parentSensor->WorldName());
                                                       ^~~~~~~~~
                                                       GetWorldName
/usr/local/Cellar/gazebo7/7.0.0-20150824-1969aea/include/gazebo-7.0/gazebo/sensors/Sensor.hh:177:27: note: 'GetWorldName' declared here
      public: std::string GetWorldName() const;
                          ^
/Users/ada/UAV/src/Firmware_cmake/Tools/sitl_gazebo/src/gazebo_lidar_plugin.cpp:85:25: error: no member named 'LaserShape' in
      'gazebo::sensors::RaySensor'; did you mean 'GetLaserShape'?
    this->parentSensor->LaserShape()->ConnectNewLaserScans(
                        ^~~~~~~~~~
                        GetLaserShape
/usr/local/Cellar/gazebo7/7.0.0-20150824-1969aea/include/gazebo-7.0/gazebo/sensors/RaySensor.hh:189:41: note: 'GetLaserShape' declared
      here
      public: physics::MultiRayShapePtr GetLaserShape() const
                                        ^
/Users/ada/UAV/src/Firmware_cmake/Tools/sitl_gazebo/src/gazebo_lidar_plugin.cpp:107:48: error: no member named 'RangeMin' in
      'gazebo::sensors::RaySensor'
  lidar_message.set_min_distance(parentSensor->RangeMin());
                                 ~~~~~~~~~~~~  ^
/Users/ada/UAV/src/Firmware_cmake/Tools/sitl_gazebo/src/gazebo_lidar_plugin.cpp:108:48: error: no member named 'RangeMax' in
      'gazebo::sensors::RaySensor'
  lidar_message.set_max_distance(parentSensor->RangeMax());
                                 ~~~~~~~~~~~~  ^
/Users/ada/UAV/src/Firmware_cmake/Tools/sitl_gazebo/src/gazebo_lidar_plugin.cpp:109:52: error: no member named 'Range' in
      'gazebo::sensors::RaySensor'
  lidar_message.set_current_distance(parentSensor->Range(0));
                                     ~~~~~~~~~~~~  ^
7 errors generated.
make[7]: *** [CMakeFiles/gazebo_lidar_plugin.dir/src/gazebo_lidar_plugin.cpp.o] Error 1
```

```
/Users/ada/UAV/src/Firmware_cmake/Tools/sitl_gazebo/src/gazebo_opticalFlow_plugin.cpp:60:5: error: no matching function for call to
      'dynamic_pointer_cast'
    std::dynamic_pointer_cast<sensors::CameraSensor>(_sensor);
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/memory:4927:1: note: candidate
      template ignored: could not match 'std::__1::shared_ptr' against 'boost::shared_ptr'
dynamic_pointer_cast(const shared_ptr<_Up>& __r) _NOEXCEPT
^
/Users/ada/UAV/src/Firmware_cmake/Tools/sitl_gazebo/src/gazebo_opticalFlow_plugin.cpp:69:9: error: no matching function for call to
      'dynamic_pointer_cast'
    if (std::dynamic_pointer_cast<sensors::DepthCameraSensor>(_sensor))
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/memory:4927:1: note: candidate
      template ignored: could not match 'std::__1::shared_ptr' against 'boost::shared_ptr'
dynamic_pointer_cast(const shared_ptr<_Up>& __r) _NOEXCEPT
^
/Users/ada/UAV/src/Firmware_cmake/Tools/sitl_gazebo/src/gazebo_opticalFlow_plugin.cpp:77:38: error: no member named 'Camera' in
      'gazebo::sensors::CameraSensor'
[... 36 lines cut ...]
      public: std::string GetImageFormat() const;
                          ^
/Users/ada/UAV/src/Firmware_cmake/Tools/sitl_gazebo/src/gazebo_opticalFlow_plugin.cpp:126:26: error: no member named 'ImageData' in
      'gazebo::rendering::Camera'; did you mean 'GetImageData'?
  _image = this->camera->ImageData(0);
                         ^~~~~~~~~
                         GetImageData
/usr/local/Cellar/gazebo7/7.0.0-20150824-1969aea/include/gazebo-7.0/gazebo/rendering/Camera.hh:365:44: note: 'GetImageData' declared
      here
      public: virtual const unsigned char *GetImageData(unsigned int i = 0);
                                           ^
/Users/ada/UAV/src/Firmware_cmake/Tools/sitl_gazebo/src/gazebo_opticalFlow_plugin.cpp:137:31: error: no member named 'RenderRate' in
      'gazebo::rendering::Camera'; did you mean 'GetRenderRate'?
  double rate = this->camera->RenderRate();
                              ^~~~~~~~~~
                              GetRenderRate
/usr/local/Cellar/gazebo7/7.0.0-20150824-1969aea/include/gazebo-7.0/gazebo/rendering/Camera.hh:107:22: note: 'GetRenderRate' declared
      here
      public: double GetRenderRate() const;
                     ^
9 errors generated.
make[7]: *** [CMakeFiles/gazebo_opticalFlow_plugin.dir/src/gazebo_opticalFlow_plugin.cpp.o] Error 1
```

Looks like I'm missing something? Or have the wrong version of something..

nicolaerosia, 2016-04-09:
Can you search your system for FindEigen.cmake or FindEigen3.cmake ?
If you have it, when you can add that path to CMAKE_MODULE_PATH

nicolaerosia, 2016-04-09:
Are you sure you have Gazebo 7?
https://github.com/PX4/sitl_gazebo/blob/master/src/gazebo_lidar_plugin.cpp#L68

AndreasAntener, 2016-04-09:
yes, 7.0.0-20150824-1969aea
/usr/local/Cellar/eigen/3.2.8/share/cmake/Modules/FindEigen3.cmake

AndreasAntener, 2016-04-09:
So Eigen will be found by adding
```
list(APPEND CMAKE_MODULE_PATH /usr/local/share/cmake/Modules)
```
which comes from `$(brew --prefix)/share/cmake/Modules`

Some info about this issue here: https://github.com/Homebrew/legacy-homebrew/issues/44937

LorenzMeier, 2016-04-09:
Does that fix the entire compilation for you?

LorenzMeier, 2016-04-09:
Fixed in 0abc792d96729544f7765e18a58a8681f0f2a96f.

closed 2016-04-09 by LorenzMeier

AndreasAntener, 2016-04-10:
@LorenzMeier no it doesn't fix the compilation issues, these are separate. I assume it works for you on Mac? Gazebo difference?

LorenzMeier, 2016-04-10:
Can you install 6.5 as per our documentation and report back if that works?

AndreasAntener, 2016-04-11:
Don't know anymore why I had 7 in the first place, did we work with that initially on Mac? Anyway, solved by reverting to 6.5.