PX4/PX4-Autopilot #4201: FW generalize Auto Land.

tubeme, 2016-04-10:
Recently I concentrated on auto take off and landing in preparation for auto take off and lending on a runway. So I stumbled on this PR https://github.com/PX4/Firmware/pull/3819 and totally agree with @dagar that there should be revision to both take off and landing. 

So this issue is for the landing part. 

Currently I use a Phantom FX61 flying wing for the experiments. So here are my problems.

If I set FW_LND_ANG = 5 as default then the slope is too high and the plane bounces from the grass and I have some minor damage from this. After it hits the ground in the bounce it turns on its motor full throttle just for a split second but enough to brake the prop the second time it lands.

I imagine if I land on a tarmac with tires the plane will bounce even more.

So I tried FW_LND_ANG = 3. I really couldn't make the mission because I had to make 1km approach.

So FW_LND_ANG = 4 was OK. The plane hit the ground in a better angle so it almost did not bounce.

In the Airliners the final approach is with 1 degree angle. So I think the landing process should be devided in phases, the same way @dagar devided the take off in PR https://github.com/PX4/Firmware/pull/3819.

I can perfectly descend with 5 deg angle then from the last mission point I can descend with 3,4 deg and bellow 10 meters I want to have a 1 degree slope until touch down.

This last phase will bellow 10 meters with 1 deg will serve as a pre flare pitch up. 

This way we will have much better dynamics for landing on runway.

Another problem I found during my experiments was that I don't see the flare working in the last phase. 

I could not understand for now what is the reason for this. Sometimes it gives good flare sometimes there is no flare... just bounce... This is dangerous for the runway landing. There should be always pretty good flare right until touch down.

This phases will apply also to landing on the green and make the landings just sliding the plane not bouncing it. 

With the current system if I want to make my approach to 1 deg slope the rules are like for the real airliners, it is pure geometry I have to make a kilometers approach and it is impractical in real life situations given the size of our planes.My mission is on a screen 1km X 0.6km  and then I have to make another 3 km for approach going outside the zone I'm flying and coming back...

What do you think?

dagar, 2016-04-10:
The current code does basically have phases, but we could probably clean it up structuring the code similar to takeoff. We'd need to really think through all the details of the design and cleanly encompass the current functionality. 
For #3819 I broke the motor mount on my Phantom so I'm waiting on a Bixler 3 to complete the generalized takeoff. If you have any thoughts there or could even help with testing I'll be able to spend some time on landing once we're happy with takeoff.

Start thinking about your ideal landing and what each phase would look like
DESCEND (optional spiral?)
APPROACH
HEADINGLIMIT
LIMITTHROTTLE
FLARE
TOUCHDOWN

jcooper21, 2016-04-11:
I may be mistaken but is this not referring to the same procedure outlined here?
http://discuss.px4.io/t/px4-flying-wing/266

tubeme, 2016-04-11:
@dagar I see that there are some phases in the code, but everything should be structured more clearly in order to be useful and practical.

Probably there is not a bad idea to move all the parameters concerning take off and landing in their corresponding tabs.

Because If I want to set parameters for the take off and landing I have to go all around tecs, l1 attitude etc to find this specific parameters. A nightmare... Like for the landing i have all the pitch control parameters, throttle  control params and the rules and altitudes for the flare in different places. I know they are organized such that they follow the logic of the code. But I really don't care about code logic, iner, outer loops, ekf etc when I'm user. For me as a user these parameters are part of my user experience, in this case I land and I take off I connect all these parameters as a meaning with take off and landing that I'm doing. I don't have to understand the code neither how it works. I want to fly.

This looks very nice. 
DESCEND (optional spiral?)
APPROACH
HEADINGLIMIT
LIMITTHROTTLE
FLARE
TOUCHDOWN

I can help with testing. Just need to repair a hybrid flying wing for a couple of days. I broke my Phantom too :) I stalled it....:(

tubeme, 2016-04-11:
@jcooper21 THis is not specifically for http://discuss.px4.io/t/px4-flying-wing/266. I'm talking about the last phases and what is the allowed pitch, roll and slope angles during this last phases. If we want a smoother touchdown for runway landing or grass landing without gear then we need smoother angles. In the current situation if we set smaller slope angles then automatically the approach becomes very long and impractical.

tubeme, 2016-04-11:
@dagar I'm building a machine for Auto landing with flaps and retractable landing gear. 

I don't know how is it implemented now, but for the take off and landing we can preset flaps. Small flaps for take off and fully extended for the landing.

My plane is with retracts. I cant seem to find a way to setup auto retracts when I'm landing? Any ideas?

I also have have a separate servo for the front wheel and cant seem to find a way to connect it such that it had a different set or PID parameters.

dagar, 2016-04-12:
I don't believe there's any support for retracts at the moment unless I've completely missed it. You could just do passthru for now and control them manually with your transmitter. Proper support tied to takeoff/landing would be a great contribution.

Add the front wheel in the mixer just like the rudder and look for FW_WR_*.

tubeme, 2016-04-12:
OK Thanks. Is there a way I could triger a servo with a mav command? If I can then I can program a custom point with mav command to trigger the servo of the gear. This could be applicable for the flaps as well. Is there auto Flaps during the landing?

dagar, 2016-04-12:
Yes flaps are deployed for auto landing.
Try MAV_CMD_DO_SET_SERVO for your landing gear, which you can do as a
waypoint in QGC.

tubeme, 2016-04-20:
They have some good ideas here especially about the flare. But I think @dagar proposal is better.

If there is no Lidar isn't it better to use GPS altitude instead of barometric altitude for the landing and flare?

closed 2018-02-21 by PX4BuildBot