PX4/PX4-Autopilot #4202: poll vs px4_poll -> unified API

bkueng, 2016-04-11:
Hi

I want to poll simultanously on an ORB topic and a serial device (in the gps thread). However these use different API's (px4_poll vs poll). At least on linux, px4_poll uses separate (user-space) file descriptors, which means the syscall poll cannot be used.
Can we combine the two API's? This would mean the px4_poll needs to use kernel file descriptors. A solution is to use pipes [0], for which poll can be used as well.
The question is also how to handle message dropping and multiple readers.
A drawback of this is a (slightly?) negative impact on performance. Any other ideas?

If this is not possible, I have to manually poll, which means sleep a short while, then check both for new data. I would not prefer this solution, and the same problem might occur at more places in the future.

[0] http://man7.org/linux/man-pages/man2/pipe.2.html

@LorenzMeier @mcharleb

bkueng, 2016-04-12:
It's actually not that much higher, we also get 3*5 RTCM msgs/s if configured at highest rate.
Most likely it will work with orb_check.

mcharleb, 2017-01-09:
@LorenzMeier Can we close this issue or is this something you want to address?

closed 2017-01-09 by LorenzMeier