PX4/PX4-Autopilot #4203: CMake Error:   HEXAGON_SDK_ROOT not set

bkueng, 2016-04-11:
Hi

The rpi2 build target seems to be broken in master:
```
$ make posix_rpi2_release
-- posix-rpi2-release
CMake Error at cmake/toolchains/Toolchain-arm-linux-gnueabihf.cmake:21 (message):
  HEXAGON_SDK_ROOT not set
Call Stack (most recent call first):
  /usr/share/cmake/Modules/CMakeDetermineSystem.cmake:98 (include)
  CMakeLists.txt:204 (project)

CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_ASM_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!
ninja -C /home/beat/px4/src/Firmware/build_posix_rpi2_release
ninja: Entering directory `/home/beat/px4/src/Firmware/build_posix_rpi2_release'
ninja: fatal: chdir to '/home/beat/px4/src/Firmware/build_posix_rpi2_release' - No such file or directory
Makefile:185: recipe for target 'posix_rpi2_release' failed
make: *** [posix_rpi2_release] Error 1
```

Can you look into this @julianoes ?

julianoes, 2016-04-11:
HEXAGON_SDK_ROOT is only needed when compiling for the Snapdragon, so that's confusing.

@bo-rc, @rosshop any ideas what is wrong here?

mhkabir, 2016-04-11:
It's not choosing the Rpi2 cross toolchain properly for some reason. Let me check.

nacho-carnicero, 2016-04-20:
Hi!

Any news on how to solve this topic? I'm stuck as well with the same error!
Thanks!
Nacho.

LorenzMeier, 2016-04-20:
@mcharleb @jywilson It would be really, really good to resolve this. As Qualcomm is handing out more boards this is affecting *a lot* of people.

mcharleb, 2016-04-20:
cmake/toolchains/Toolchain-arm-linux-gnueabihf.cmake should not need to be used for Snapdragon Flight builds. The Snapdragon Flight builds for the apps proc should use: cmake_hexagon/toolchain/Toolchain-arm-linux-gnueabihf.cmake

That way, cmake/toolchains/Toolchain-arm-linux-gnueabihf.cmake does not need to depend on HEXAGON_SDK_ROOT

nacho-carnicero, 2016-04-20:
Thanks for answering, but the problem here has to do with Raspberry Pi builds, and not with Snapdragon Flight builds. According to @mhkabir the build target is not choosing the RPI2 toolchain, any hints where this can come from?

mhkabir, 2016-04-29:
OK, found the problem. Fix coming up.

mhkabir, 2016-04-29:
All interested, please see #4404.

julianoes, 2016-07-13:
I'm pretty sure this is fixed now, closing.

closed 2016-07-13 by julianoes