PX4/PX4-Autopilot #4212: Running multiple SITL instances

marianobianchi, 2016-04-11:
I am working on a project that needs to manage multiple UAVs simultaneously. In order to test our code before using it with real UAVs, we want to simulate our environment using px4 SITL.

Is there any way to simulate more than one UAV using px4 SITL? How can i do it?

marianobianchi, 2016-06-09:
@AndreasAntener Is this something similar to #4578 ??

cynthiamoon, 2016-11-15:
how to insert another PX4 model on gazebo?

julianoes, 2016-11-16:
@marianobianchi has this been resolved or is it still an issue?

@cynthiamoon what exactly do you want to do?

cynthiamoon, 2016-11-16:
@julianoes i read this.[https://github.com/PX4/sitl_gazebo/pull/30](url).i want to run multiple autopilots  in one SITL gazebo.i know different physical objects in Gazebo must have different names.i don't  know how to add another autopilots on one SITL gazebo.

julianoes, 2016-11-16:
I don't know how this works but a pointer is here: https://github.com/PX4/Firmware/blob/master/Tools/sitl_multiple_run.sh

The point is you also need the PX4 simulator, etc. to run for every model, just adding a model in Gazebo won't do it.