PX4/PX4-Autopilot #4213: Pixhawk not recognized when reconnected to the computer after arming it

nacho-carnicero, 2016-04-11:
Hello, 

I am having some problems with re-connecting to the Pixhawk with my computer (both with QGC or with the terminal) after having armed and unarmed it. Here is what I have done:
I flashed the PX4 software from the repository directly through the terminal (I'm using Mac with El Capitan). I then connected the Pixhawk to a battery, connected to QGC through the 3DR telemetry, did all the sensor calibration and armed it in order to perform some tests in a test-bench (the Pixhawk is only connected to the switch, a battery, a RC controller and a servo). Once I carried all the tests I unarmed it with the safety switch and unplugged the battery. Up to this point everything went fine. 

But now, if I try to reconnect the Pixhawk to the computer, no matter if through the USB cable or the 3DR telemetry the computer does not recognize it. No serial port is listed and therefore I can not connect to the board through QGC nor reflash the firmware through the terminal because the computer does not see any serial port. The strange thing is that if I use the other USB port I have on my laptop it recognizes it, but if I redo the same process I reach the same result but now changing the USB port does not work and the only thing that fixes the problem is restarting the computer.

Having to restart the computer every two times that I want to connect to the Pixhawk is really annoying, does anyone have had the same problem? Here is what I am using:

Mac with OS X El Capitan
Pixhawk board with USB cable or 3DR telemetry
QGC v2.9.4
The firmware from the repository cloned one week ago

Thanks for the help, 

Nacho.

suric, 2016-04-11:
I have the same problem. Try to erase the sd-card!

nacho-carnicero, 2016-04-12:
Hey thanks for the reply!
I tried that and today it seems to be working better, I'll keep you posted!

nacho-carnicero, 2016-04-12:
Ok so I still have the problem. I thought that erasing the SD card would have solved it but it didn't. So same as last time, I plug the Pixhawk to QGC, perform some tests (in a Testbench) and when I unarm it the computer does not recognize it anymore. Well indeed it recognizes it but no serial port is opened and therefore I cannot connect to the board. 
This is what the terminal shows me if I ask it to list the dev/cu.* detected (the port corresponding to the Pixhawk is /dev/cu.usbmodem1 or /dev/tty.usbmodem1)
[image]
But this is what my OS system info shows 
[image]
It recognizes the Pixhawk board but there is no serial port created!

If I do the same operation with the other USB port here is what the terminal shows:
[image]
And here is what my OS system info shows:
[image]
So now my laptop recognizes the Pixhawk board and creates the port in order to connect to it. 
But if I arm the Pixhawk and perform some tests after some point this USB port won't work neither and therefore the only thing I can do is restart the computer.

Does any of you know why this happens? Thanks for the help, 

Nacho

nacho-carnicero, 2016-04-12:
Ok I think I solved the mystery. Starting from the OS X Mavericks release Apple decided to develop its own FTDI driver. If you have the official FTDI driver you should remove it since there appear to be some compatibility issues (for further information you can go to https://developer.apple.com/library/mac/technotes/tn2315/_index.html).

I have tried to remove it and it works (for now)! To remove it: 
Open the terminal and type:
`sudo rm -r /System/Library/ExtensionsFTDIUSBSerialDriver.kext`
It is likely that the drivers are not located in the /System/Library/Extensions but in /Library/Extensions (or both like in my case), therefore type in the terminal:
`sudo rm -r /Library/Extensions/FTDIUSBSerialDriver.kext`
If you are having troubles performing these actions try restarting the computer before since if the FTDI driver has been already launched you won't be able to remove it.

And that's it, then no more problems where the Pixhawk is listed as connected to the computer but no port is opened to connect to it!

Nacho

closed 2016-04-12 by nacho-carnicero

Jacob12138xieyuan, 2018-09-17:
I have same problem

dagar, 2018-09-17:
@Jacob12138xieyuan could you please open a new issue with your specific info? This one is over 2 years old and closed.