PX4/PX4-Autopilot #4214: Simplify calibration process for end user

jgoppert, 2016-04-11:
It would be good to align our calibration procedure to some of the other autopilots out there like dji and parrot. In general it takes too long for end users right now.

dagar, 2016-04-11:
On the call I believe you said about using SVD, but that wouldn't be feasible for a pixhawk. Could we offload that to QGC?

jgoppert, 2016-04-11:
Yeah, we could use SVD on QGC, but I think if you assume the ellipsoid is aligned with the primary axis of the sensors that the math gets simplified, and you might not need SVD. We would still probably need a pseudo-inverse though, but I think someone must have figured out an easy way to do this on microprocessors.

dagar, 2017-04-28:
I think this is covered by https://github.com/PX4/Firmware/issues/6472 and https://github.com/PX4/Firmware/issues/6465

closed 2017-04-28 by dagar